arduino/ArduinoCore-mbed #184: Example for Pico PIO

joachimhs, 2021-04-10:
Hello,

It is great that support for the Pi Pico has landed in the latest version! I am wondering though. Are there any examples that show how to program the PIO through either Arduino IDE or through arduino-cli?

facchinm, 2021-04-12:
Hi @joachimhs ,
we still didn't integrate PIO files compilation into the toolchain, but library developers can compile via `pioasm` and then add the resulting `.h` to any library (we used PIO for PDM support for example https://github.com/arduino/ArduinoCore-mbed/tree/master/libraries/PDM/src/rp2040).
We are still thinking about a user friendly way to expose the raw functionality, but in the meantime just providing the "precompiled" version wrapped via high level APIs should be suitable for end users :wink:

joachimhs, 2021-04-13:
@facchinm Thank you for your reply! I did find this clone of the Adafruit NeoPixel library that uses the PIO to interface with the LEDs. I will look into compiling with pioasm as an option. (https://github.com/Bodmer/Adafruit_NeoPixel)

tierneytim, 2021-06-09:
Thanks for the hard work on this guys. I much prefer the Arduino core to the official SDK (I work on windows). To make working with the PIO easier could a compiled pioasm be distributed (or built at installation)? Otherwise, the end-user has to install CMake, MinGW, and the official SDK (or is there an easier way?). Then they need to compile the pioasm.exe. This seems excessive considering the PIO is such a strong selling point of this microcontroller.

GregWoods, 2021-08-28:
I agree. Personally, if it wasn't for PIO I'd just use an STM32F4.
PIO is **_the_** selling point of the RP2040 !
So, yes, I would love to see built in support for compiling and uploading pio asm code.

Bodmer, 2022-01-06:
There is a nice online pioasm here:
https://wokwi.com/tools/pioasm

facchinm, 2022-01-11:
@joachimhs @GregWoods @tierneytim can you give a spin to https://github.com/arduino/rp2040tools/releases/tag/1.0.5-rc1 and report if the tools work fine in your environment? So we can make a new release of the core that also includes precompiled `pioasm` :wink: