arduino/ArduinoCore-mbed #185: error with RPI PICO RP2040

759401524, 2021-04-12:
```
.arduino15/packages/arduino/hardware/mbed_rp2040/2.0.0/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_stdlib/include/pico/stdlib.h:11:10: fatal error: pico/stdio.h: No such file or directory
 #include "pico/stdio.h"
          ^~~~~~~~~~~~~~
compilation terminated.
exit status 1
```
There seems to be something wrong with pico/stdlib

facchinm, 2021-04-12:
Hi  @759401524 ,
can you please post the full verbose compile output and your installation specifications (OS, IDE version)? Thanks

759401524, 2021-04-13:
```
Arduino:1.8.13 (Linux), 开发板："Raspberry Pi Pico"

/home/mulong/Videos/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /home/mulong/Videos/arduino-1.8.13/hardware -hardware /home/mulong/.arduino15/packages -tools /home/mulong/Videos/arduino-1.8.13/tools-builder -tools /home/mulong/Videos/arduino-1.8.13/hardware/tools/avr -tools /home/mulong/.arduino15/packages -built-in-libraries /home/mulong/Videos/arduino-1.8.13/libraries -libraries /home/mulong/Arduino/libraries -fqbn=arduino:mbed_rp2040:pico -ide-version=10813 -build-path /tmp/arduino_build_856436 -warnings=default -build-cache /tmp/arduino_cache_630761 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=/home/mulong/.arduino15/packages/arduino/tools/dfu-util/0.10.0-arduino1 -prefs=runtime.tools.dfu-util-0.10.0-arduino1.path=/home/mulong/.arduino15/packages/arduino/tools/dfu-util/0.10.0-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/mulong/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/home/mulong/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.rp2040tools.path=/home/mulong/.arduino15/packages/arduino/tools/rp2040tools/1.0.2 -prefs=runtime.tools.rp2040tools-1.0.2.path=/home/mulong/.arduino15/packages/arduino/tools/rp2040tools/1.0.2 -prefs=runtime.tools.bossac.path=/home/mulong/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=/home/mulong/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino2 -prefs=runtime.tools.openocd.path=/home/mulong/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino1 -prefs=runtime.tools.openocd-0.11.0-arduino1.path=/home/mulong/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino1 -verbose /tmp/arduino_modified_sketch_523486/Blink.ino
/home/mulong/Videos/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /home/mulong/Videos/arduino-1.8.13/hardware -hardware /home/mulong/.arduino15/packages -tools /home/mulong/Videos/arduino-1.8.13/tools-builder -tools /home/mulong/Videos/arduino-1.8.13/hardware/tools/avr -tools /home/mulong/.arduino15/packages -built-in-libraries /home/mulong/Videos/arduino-1.8.13/libraries -libraries /home/mulong/Arduino/libraries -fqbn=arduino:mbed_rp2040:pico -ide-version=10813 -build-path /tmp/arduino_build_856436 -warnings=default -build-cache /tmp/arduino_cache_630761 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=/home/mulong/.arduino15/packages/arduino/tools/dfu-util/0.10.0-arduino1 -prefs=runtime.tools.dfu-util-0.10.0-arduino1.path=/home/mulong/.arduino15/packages/arduino/tools/dfu-util/0.10.0-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/mulong/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/home/mulong/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.rp2040tools.path=/home/mulong/.arduino15/packages/arduino/tools/rp2040tools/1.0.2 -prefs=runtime.tools.rp2040tools-1.0.2.path=/home/mulong/.arduino15/packages/arduino/tools/rp2040tools/1.0.2 -prefs=runtime.tools.bossac.path=/home/mulong/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=/home/mulong/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino2 -prefs=runtime.tools.openocd.path=/home/mulong/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino1 -prefs=runtime.tools.openocd-0.11.0-arduino1.path=/home/mulong/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino1 -verbose /tmp/arduino_modified_sketch_523486/Blink.ino
Using board 'pico' from platform in folder: /home/mulong/.arduino15/packages/arduino/hardware/mbed_rp2040/2.0.0
Using core 'arduino' from platform in folder: /home/mulong/.arduino15/packages/arduino/hardware/mbed_rp2040/2.0.0
Detecting libraries used...
/home/mulong/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g3 -nostdlib @/home/mulong/.arduino15/packages/arduino/hardware/mbed_rp2040/2.0.0/variants/RASPBERRY_PI_PICO/defines.txt @/home/mulong/.arduino15/packages/arduino/hardware/mbed_rp2040/2.0.0/variants/RASPBERRY_PI_PICO/cxxflags.txt -DARDUINO_ARCH_RP2040 -mcpu=cortex-m0plus -w -x c++ -E -CC -DARDUINO=10813 -DARDUINO_RASPBERRY_PI_PICO -DARDUINO_ARCH_MBED_RP2040 -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 -I/home/mulong/.arduino15/packages/arduino/hardware/mbed_rp2040/2.0.0/cores/arduino -I/home/mulong/.arduino15/packages/arduino/hardware/mbed_rp2040/2.0.0/variants/RASPBERRY_PI_PICO -I/home/mulong/.arduino15/packages/arduino/hardware/mbed_rp2040/2.0.0/cores/arduino/api/deprecated -I/home/mulong/.arduino15/packages/arduino/hardware/mbed_rp2040/2.0.0/cores/arduino/api/deprecated-avr-comp -iprefix/home/mulong/.arduino15/packages/arduino/hardware/mbed_rp2040/2.0.0/cores/arduino @/home/mulong/.arduino15/packages/arduino/hardware/mbed_rp2040/2.0.0/variants/RASPBERRY_PI_PICO/includes.txt /tmp/arduino_build_856436/sketch/Blink.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for pico/stdio.h: []
ResolveLibrary(pico/stdio.h)
  -> candidates: []
In file included from /tmp/arduino_modified_sketch_523486/Blink.ino:24:0:
/home/mulong/.arduino15/packages/arduino/hardware/mbed_rp2040/2.0.0/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_stdlib/include/pico/stdlib.h:11:10: fatal error: pico/stdio.h: No such file or directory
 #include "pico/stdio.h"
          ^~~~~~~~~~~~~~
compilation terminated.
exit status 1
为开发板 Raspberry Pi Pico 编译时出错。

```

759401524, 2021-04-13:
This is my code
```
#include "pico/stdlib.h"

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

facchinm, 2021-04-13:
Is there any particular function you need to use from `pico/stdlib.h` ? Otherwise, the include is unnecessary, since mbed environment provides all the needed `stdlib` APIs.

759401524, 2021-04-13:
Ok, I know. Thanks

closed 2021-04-15 by 759401524