arduino/ArduinoCore-mbed #187: PulseIn function not working with RP2040

Abuthala, 2021-04-12:
I downloaded the Arduino ide to use an hc sr04 ultrasonic sensor to measure distance. In the code, I used the pulseIn function to measure the pulse on time of the echo pin. but the function seems to not be defined. The error is "sketch\sketch_apr12b.ino.cpp.o: In function `loop':
D:\ArduinoCode\sketch_apr12b/sketch_apr12b.ino:28: undefined reference to `pulseIn'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Raspberry Pi Pico."
The code is as follows: 

"int trig=15;
int echo=14;
void setup()
{
  pinMode(trig,OUTPUT);
  pinMode(echo,INPUT);
  Serial.begin(9600);
}
void loop()
{
  digitalWrite(trig,LOW);
  delayMicroseconds(2);
  digitalWrite(trig,HIGH);
  delayMicroseconds(10);
  digitalWrite(trig,LOW);
  float duration=pulseIn(echo,HIGH);
  float distance=0.02*duration;
  Serial.println(distance);
  delay(2000);
}"

facchinm, 2021-04-19:
Should be fixed by https://github.com/arduino/ArduinoCore-mbed/pull/190/commits/f5ddfa85431fa7ae391118cbb48c986687dba497

per1234, 2021-06-14:
@facchinm can this be closed now?

closed 2021-06-15 by facchinm