arduino/ArduinoCore-mbed #189: Raspberry Pico on macOS

sstaub, 2021-04-13:
macOS 11.2.3
After uploading to the Raspberry Pico board, I get a message that the board is unmounted incorrectly.
This is specific to macOS, it seems that uploading works over mass storage, that cause the problem.

facchinm, 2021-04-13:
Hi @sstaub ,
the core doesn't use the mass storage interface for uploading but it gets exposed by default; this patch https://github.com/arduino/ArduinoCore-mbed/pull/190/commits/91d896f22701dd1588e6c3b08988049c89acd927 should avoid that so during normal flow the notification won't appear.

closed 2021-04-13 by facchinm

sstaub, 2021-05-28:
@facchinm When did you merge the solution?

facchinm, 2021-05-28:
I had to revert the patch here https://github.com/arduino/ArduinoCore-mbed/commit/d2942e207a5b71ec564cd1fe9eacf2b6a88de5a2 since it would have broken all Windows "pure" Pico installations (micropython and so on). If there's any Windows drivers guru out there that knows how to handle the situation I'd be glad to turn it back on :slightly_smiling_face:

sstaub, 2021-05-28:
We need to know which operation system is used, maybe there is a macro for that in gcc.

sstaub, 2021-05-28:
Ok, I understood. In the moment I need to update manually with every version update.