arduino/ArduinoCore-mbed #191: Unable to upload sketch to raspberry pi pico

Tejas-Singh-py, 2021-04-14:
Hello,
I have just installed Arduino IDE on my raspberry pi 3 to program my pico(RP2040). I opened the blink example but after it compiled it showed an error while uploading (The pico was in bootloader mode). What should I do now. Is it because of the ARM compatibility issue?
**I used the official version, not @earlephilhower 's version*'*
P.S. This is my first time with Arduino IDE. Sorry for (if any) typos.

Youtube:- Tejas Singh YT

facchinm, 2021-04-14:
Hi @Tejas-Singh-py ,
the problem might be due to missing `udev` rules (we don't use mass storage upload but the other interface exposed by the bootloader). Try running this (https://github.com/arduino/ArduinoCore-mbed/blob/master/post_install.sh) script as root and then retry the upload (disconnecting and reconnectiong the Pico might help).
If this doesn't help, for further debugging please enable verbose upload from File/Preferences and paste here the output, thanks!

Tejas-Singh-py, 2021-04-14:
Thanks, it worked 
please check out my youtube channel https://www.youtube.com/channel/UCCsEfbaCHPjM0Dhx69TrJPw

closed 2021-04-14 by Tejas-Singh-py

avielbr, 2022-05-02:
@facchinm This worked for me as well to solve the same problem. Thanks!

29chandu, 2022-09-14:
@facchinm thanks for the solution

Pomax, 2022-09-23:
Any way to run this on Windows?