arduino/ArduinoCore-mbed #193: Sending more than 63 characters causes Serial to hang

technoblogy, 2021-04-14:
With this test program:
````
void setup() {
  Serial.begin(9600);
}

void loop (void) {
  if (Serial.available()) Serial.print((char)Serial.read());
}
````
if you repeatedly paste this 62-character string into the Arduino IDE Serial Monitor input field and press Return it gets echoed fine:

    $0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXY

However if you use this string of 63 characters it immediately hangs up:

    $0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ

I'm using the core **Arduino Mbed OS RP2040 Boards** 2.0.0 on a MacBook Pro with Arduino IDE 1.8.13.

facchinm, 2021-04-16:
Hi @technoblogy ,
thanks for reporting. On Linux there's no such issue (just tested), so I'd ask @sebromero or @alranel to test the issue on their shiny Macbooks :wink:

sebromero, 2021-04-22:
@facchinm Confirmed. Looks like it's time to solder the SWD headers...

technoblogy, 2021-05-09:
Any progress on tracking this down?

technoblogy, 2021-05-18:
Thanks! That's enabled me to port my Lisp interpreter, uLisp, to the Raspberry Pi Pico;

http://www.ulisp.com/show?3KN3

alranel, 2021-08-02:
@technoblogy Consider mentioning the [Arduino Nano RP2040 Connect](https://store.arduino.cc/nano-rp2040-connect) board in your tutorial as a way to support our work on this open source software :)

technoblogy, 2021-08-02:
OK, thanks for the suggestion, I'll get hold of an Arduino Nano RP2040 Connect board with the aim of adding that to my tutorial.