arduino/ArduinoCore-mbed #197: printf functionality

sstaub, 2021-04-17:
Support for `printf` should added.
It should work without doing with Serial, like when using Mbed, e.g. `printf("hello");` for debug purpose on USB connection to the Serial monitor.
`printf` functionality should also added when using Serial or LiquidCrystal and other libraries.

Abuthala, 2021-04-17:
Did mbed boards had that functionality before? From my knowledge, Arduino doesn't support printf. To print something, u need to use serial communication between Arduino and the pc/mac. Then,u need to open up the serial monitor to see the printed message.

sstaub, 2021-04-17:
Yes, to use printf for direct output to serial monitor is meanwhile standard (I think since Mbed 6) behavior on Mbed.
The limitation for printf is caused of the limited flash space of AVR boards.

facchinm, 2021-04-19:
`printf` works on mbed since it outputs to the default UART (which is usually exported as micro USB by a CMSIS-DAP programmer).
You can indeed use it and it will print on Serial1 (default UART, pin 1 for Nano33BLE, pin 13 for Portenta), or our can use the `REDIRECT_STDOUT_TO` macro to redirect printf to the SerialUSB object (see here https://github.com/arduino/ArduinoCore-mbed/issues/174#issuecomment-814027679 for details).

closed 2021-04-19 by facchinm