arduino/ArduinoCore-mbed #200: Bug: Common Arduino Ethernet examples not working

hpssjellis, 2021-04-20:
Trying any of the common Arduino Ethernet examples I get this error:

```
#error "Either IPv4 or IPv6 must be preferred."
  

```

I have been able to get an mbed example working

```
/*
 * Copyright (c) 2006-2020 Arm Limited and affiliates.
 * SPDX-License-Identifier: Apache-2.0
 */
#include "mbed.h"
#include "EthernetInterface.h"

// Network interface
EthernetInterface net;

// Socket demo
int main()
{
    // Bring up the ethernet interface
    printf("Ethernet socket example\n");
    net.connect();

    // Show the network address
    SocketAddress a;
    net.get_ip_address(&a);
    printf("IP address: %s\n", a.get_ip_address() ? a.get_ip_address() : "None");

    // Open a socket on the network interface, and create a TCP connection to mbed.org
    TCPSocket socket;
    socket.open(&net);

    net.gethostbyname("ifconfig.io", &a);
    a.set_port(80);
    socket.connect(a);
    // Send a simple http request
    char sbuffer[] = "GET / HTTP/1.1\r\nHost: ifconfig.io\r\n\r\n";
    int scount = socket.send(sbuffer, sizeof sbuffer);
    printf("sent %d [%.*s]\n", scount, strstr(sbuffer, "\r\n") - sbuffer, sbuffer);

    // Recieve a simple http response and print out the response line
    char rbuffer[64];
    int rcount = socket.recv(rbuffer, sizeof rbuffer);
    printf("recv %d [%.*s]\n", rcount, strstr(rbuffer, "\r\n") - rbuffer, rbuffer);

    // Close the socket to return its memory and bring down the network interface
    socket.close();

    // Bring down the ethernet interface
    net.disconnect();
    printf("Done\n");
}

```

but not sure how to convert that to a simple webServer or ChatServer. With both the Portenta breakout board and ethernet vision shield out it would be good to have a few more ethernet examples.

facchinm, 2021-04-21:
Hi Jeremy,
the problem is, not unexpectedly, case insensitive filesystems :slightly_smiling_face: 
Basically `Ethernet.h` is not being discovered since `ethernet.h` is in the include search path (from lwip).
To overcome this, I prepared this patch https://github.com/arduino/ArduinoCore-mbed/commit/c271a1795ecae6987ae08a1ae9f7a72dbe195f8c . If you apply it and include `Portenta_Ethernet.h` in your sketch (before including `Ethernet.h`) than the right library gets selected .
The whole thing is not ideal but can be used as a temporary workaround until we find something better.
@manchoz

pennam, 2021-04-21:
To compile [webserver example](https://www.arduino.cc/en/Tutorial/LibraryExamples/WebServer) you need also to apply this https://github.com/arduino/ArduinoCore-mbed/pull/202/commits/f7acf0b4c27109d1ea9d2adcef008a7bf9199a83 and change 

```
// start the Ethernet connection and the server:
Ethernet.begin(mac, ip);
```
to  

```
// start the Ethernet connection and the server:
Ethernet.begin(mac);
```

because manual configuration is not implemented

hpssjellis, 2021-04-21:
@pennam, why do I even try using Windows, compiles fine on ubuntu.  

However the webserver is having a hardware issue. Do you have a Portenta Ethernet Vision Shield?  Curious if the webserver code works for you. I made your change, but this line kills it.

If I bloke it out, the IP becomes 0.0.0.0. Suggestions.

```
  if (Ethernet.hardwareStatus() == EthernetNoHardware) {
    Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
    while (true) {
      delay(1); // do nothing, no point running without Ethernet hardware
    }
  }

```

I think the Portenta may have different SPI settings than the MKRWAN. This might be the issue
```

 Circuit:
 * Ethernet shield attached to pins 10, 11, 12, 13
...

#include <SPI.h>
...

void setup() {
  // You can use Ethernet.init(pin) to configure the CS pin
  //Ethernet.init(10);  // Most Arduino shields
  Ethernet.init(5);   // MKR ETH shield

```

Any suggestions?

P.S. Where is the Ethernet.h file, all I can find in my core is the ethernet.h file.

pennam, 2021-04-21:
> To compile [webserver example](https://www.arduino.cc/en/Tutorial/LibraryExamples/WebServer) you need also to apply this [f7acf0b](https://github.com/arduino/ArduinoCore-mbed/commit/f7acf0b4c27109d1ea9d2adcef008a7bf9199a83) and change
> 
> ```
> // start the Ethernet connection and the server:
> Ethernet.begin(mac, ip);
> ```
> 
> to
> 
> ```
> // start the Ethernet connection and the server:
> Ethernet.begin(mac);
> ```
> 
> because manual configuration is not implemented

Not really true, you need to take https://github.com/arduino/ArduinoCore-mbed/commit/f7acf0b4c27109d1ea9d2adcef008a7bf9199a83 only if you move to master branch HEAD otherwise if you are using https://github.com/arduino/ArduinoCore-mbed/releases/tag/2.0.0 you just need https://github.com/arduino/ArduinoCore-mbed/commit/c271a1795ecae6987ae08a1ae9f7a72dbe195f8c

pennam, 2021-04-21:
> @pennam, why do I even try using Windows, compiles fine on ubuntu.
>
> However the webserver is having a hardware issue. Do you have a Portenta Ethernet Vision Shield? Curious if the webserver code [...]

yes, in this case #include <SPI.h> is not needed and you don't neither have to call Ethernet.init()

pennam, 2021-04-21:
This is the sketch code i'm using

```
#include <Portenta_Ethernet.h>
#include <Ethernet.h>

// Enter a MAC address and IP address for your controller below.
// The IP address will be dependent on your local network:
byte mac[] = {
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED
};
IPAddress ip(192, 168, 1, 177);

// Initialize the Ethernet server library
// with the IP address and port you want to use
// (port 80 is default for HTTP):
EthernetServer server(80);

void setup() {
  // You can use Ethernet.init(pin) to configure the CS pin
  //Ethernet.init(10);  // Most Arduino shields
  //Ethernet.init(5);   // MKR ETH shield
  //Ethernet.init(0);   // Teensy 2.0
  //Ethernet.init(20);  // Teensy++ 2.0
  //Ethernet.init(15);  // ESP8266 with Adafruit Featherwing Ethernet
  //Ethernet.init(33);  // ESP32 with Adafruit Featherwing Ethernet

  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  Serial.println("Ethernet WebServer Example");

  // start the Ethernet connection and the server:
  Ethernet.begin(mac);

  // Check for Ethernet hardware present
  if (Ethernet.hardwareStatus() == EthernetNoHardware) {
    Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
    while (true) {
      delay(1); // do nothing, no point running without Ethernet hardware
    }
  }
  if (Ethernet.linkStatus() == LinkOFF) {
    Serial.println("Ethernet cable is not connected.");
  }

  // start the server
  server.begin();
  Serial.print("server is at ");
  Serial.println(Ethernet.localIP());
}

void loop() {
  // listen for incoming clients
  EthernetClient client = server.available();
  if (client) {
    Serial.println("new client");
    // an http request ends with a blank line
    boolean currentLineIsBlank = true;
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        Serial.write(c);
        // if you've gotten to the end of the line (received a newline
        // character) and the line is blank, the http request has ended,
        // so you can send a reply
        if (c == '\n' && currentLineIsBlank) {
          // send a standard http response header
          client.println("HTTP/1.1 200 OK");
          client.println("Content-Type: text/html");
          client.println("Connection: close");  // the connection will be closed after completion of the response
          client.println("Refresh: 5");  // refresh the page automatically every 5 sec
          client.println();
          client.println("<!DOCTYPE HTML>");
          client.println("<html>");
          // output the value of each analog input pin
          for (int analogChannel = 0; analogChannel < 6; analogChannel++) {
            int sensorReading = analogRead(analogChannel);
            client.print("analog input ");
            client.print(analogChannel);
            client.print(" is ");
            client.print(sensorReading);
            client.println("<br />");
          }
          client.println("</html>");
          break;
        }
        if (c == '\n') {
          // you're starting a new line
          currentLineIsBlank = true;
        } else if (c != '\r') {
          // you've gotten a character on the current line
          currentLineIsBlank = false;
        }
      }
    }
    // give the web browser time to receive the data
    delay(1);
    // close the connection:
    client.stop();
    Serial.println("client disconnected");
  }
}
```

hpssjellis, 2021-04-21:
Awesome, all good now. Hopefully those patches will be in the next Portenta release. I tested it on windows and it worked great.  @pennam and @facchinm presently my favourite dev's.

hpssjellis, 2021-05-19:
Has anyone tested both PortentaH7 WiFi and Ethernet on Windows and Linux with the new 2.1.0 Arduino-mbed core? Not sure if I need better examples or the code still has issues. I can get @pennam code above working on Linux, but it only loads one page and nothing else. The whole Portenta Wifi and Ethernet is sorely low on working examples.

pennam, 2021-05-21:
@hpssjellis, you are right there is an issue related to ethernet connection and reading analog input 1, most probably is because pins PA1 and PA1_C can be internally connected, but i need  some time to investigate. In the meanwhile if you skip reading analogChannel 1 also the above sketch posted by me should work.

hpssjellis, 2021-05-22:
@penman, yes I was starting to figure that out. For a while it looked like A0, A6 and A3 were also causing issues but then I could not confirm it. Glad you caught it. We can close this issue now. Good work.

hpssjellis, 2021-06-22:
this was solved with the core version 2.1.0

closed 2021-06-22 by hpssjellis