arduino/ArduinoCore-mbed #205: EEPROM emulation

sstaub, 2021-04-22:
I'm missing an EEPROM emulation like on STM32duino

facchinm, 2021-04-26:
Hi @sstaub ,
using mbed's blockdevice infrastructure you can create very easily something much more powerful and reliable (or even a library that wraps EEPROM APIs).
For example, to store some data in a Pico, you can write something like this
```
#include "FlashIAPBlockDevice.h"
#include "KVStore.h"
#include "TDBStore.h"

// 512KB block device, starting 1MB inside the flash
FlashIAPBlockDevice bd(XIP_BASE + 1024*1024, 1024*512);
mbed::TDBStore eeprom(&bd);

uint8_t ram_buffer[50];
char* data = "quitealongstring";

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial); 
  eeprom.init();
  mbed::KVStore::info_t info;
  if (eeprom.get_info("mydata", &info) != MBED_ERROR_ITEM_NOT_FOUND) {
    Serial.println("Key size: " + String(info.size));
    eeprom.get("mydata", ram_buffer, info.size);
    Serial.print("Data: ");
    Serial.println((char*)ram_buffer);
  } else {
    Serial.println("Setting data");
    eeprom.reset();
    eeprom.set("mydata", data, strlen(data), 0);
  }
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
Or even use filesystems; you can  find more examples in this thread https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/16

sstaub, 2021-04-26:
Thank you for the advices.

rmlearney-digicatapult, 2021-04-27:
Problem I've found with KVSTORE is that it doesn't support integers, only strings (more specifically char*). This is a real problem when wanting to store any data in a non-volatile manner.

Are there any lightweight solutions to this?

technoblogy, 2021-05-18:
@rmlearney-digicatapult I'm not an expert on this, but as no one has answered I'll try and help.

You can use the Block Device functions with different types of variable, not just char*. For example if you've got:

    uint32_t imagesize;
    char SymbolTable[SYMBOLTABLESIZE];
    uint32_t Workspace[WORKSPACESIZE];

you can save them all with the sequence:

    eeprom.set("size", &imagesize, 4, 0);
    eeprom.set("symb", &SymbolTable, SYMBOLTABLESIZE, 0);
    eeprom.set("work", &Workspace, WORKSPACESIZE*4, 0);

and load them back in with:

    eeprom.get("size", &imagesize, 4);
    eeprom.get("symb", &SymbolTable, SYMBOLTABLESIZE);
    eeprom.get("work", &Workspace, WORKSPACESIZE*4);

The sizes all need to be in bytes. I'm not sure how long the keys can be; I've chosen four characters.

lyomov, 2024-02-18:
Hi @facchinm,

I was studying and try to implement a code you suggested at [Implementation of files read/write on flash memory.] discussion.
I like it and find it a very helpful - thanks and congratulations. 

Employing your example code, given there, in application I try to create, I got some error codes, like -22, 33. 

My question is: **Where I can find the meaning of the code errors?**

Sorry I am asking here, since the mentioned discussion is closed!

Thank you!

manchoz, 2024-02-19:
Hey @lyomov,
Look at [Creating a Flash-Optimized Key-Value Store](https://docs.arduino.cc/tutorials/portenta-h7/flash-optimized-key-value-store).
In this example, there are a few expedients that might help you solve your issues.