arduino/ArduinoCore-mbed #206: Using the second processor on a pico without rtos overheads

stuartdd, 2021-04-23:
I know this is not really an issue other than I cannot find any specific examples or docs that cover this. 

All of the discussions I have looked at so far use Scheduler, Thread or other rtos resources. 

What I would really like to happen is to activate loop2() with the second processor running that code with the same functionality, memory access and priority as loop().

I really don't want the overheads of a scheduler, managed threads or a real time operating system. I want two native concurrent processes.

Is this (or something close to it) implemented in the Arduino mbed core API.

dsyleixa, 2021-04-24:
I have to agree, it would be very handsome if one could use a simplified, (more or less) POSIX-like
sched_setaffinity( threadID, cpu_number );
so having e.g.
thread t1( blink);
thread t2( foo);
thread t3( foo);
thread t4( sthelse);
one  might set then
// assign threads to cpu number 0:
sched_setaffinity(t1, 0);  
sched_setaffinity(t2, 0);
// assign threads to cpu number 1:
sched_setaffinity(t3, 1);
sched_setaffinity(t4, 1);

dsyleixa, 2021-04-29:
I have to agree, it would be very handsome if one could use 
- from outside of each std::thread a simplified, (more or less) POSIX-like
pthread_setaffinity_np( thread.native_handle(), cpu_number ); // originally for LINUX a cpu_set is passed
- frm inside of either single thread
sched_setaffinity( 0, cpu_number );  // originally  for LINUX a cpu_set is passed

so having e.g.
std::thread t1( blink);
std::thread t2( foo);
std::thread t3( bas);
std::thread t4( sthelse);

one  might set then e.g. from main():

// assign threads to cpu number 0:
pthread_setaffinity_np(t1.native_handle(), 0);  
pthread_setaffinity_np(t2.native_handle(), 0);

// assign threads to cpu number 1:
pthread_setaffinity_np(t3.native_handle(), 1);
pthread_setaffinity_np(t4.native_handle(), 1);