arduino/ArduinoCore-mbed #207: Uploading to Pico - missing rp2040load file

santaimpersonator, 2021-04-23:
I installed the Arduino Core for the RP2040 through the board manager, but I am having issues uploading the blink sketch onto the board.

- I have installed the latest version available through the board manager:
    [image]

- This is the current error output. The code compiles fine, there seems to be a `rp2040load` file/tool missing:
    ```
    Sketch uses 80072 bytes (0%) of program storage space. Maximum is 16777216 bytes.
    
    Global variables use 52920 bytes (19%) of dynamic memory, leaving 217416 bytes for local variables. Maximum is 270336 bytes.
    
    java.io.IOException: Cannot run program "C:\Users\<username>\Documents\ArduinoData\packages\arduino\tools\rp2040tools\1.0.2/rp2040load": CreateProcess error=2, The system cannot find the file specified
	    at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	    at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	    at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	    at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	    at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	    at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	    at processing.app.SketchController.upload(SketchController.java:732)
	    at processing.app.SketchController.exportApplet(SketchController.java:703)
	    at processing.app.Editor$UploadHandler.run(Editor.java:2055)
	    at java.lang.Thread.run(Thread.java:748)
    
    Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	    at java.lang.ProcessImpl.create(Native Method)
	    at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	    at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	    at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	    ... 9 more
    
    An error occurred while uploading the sketch
    ```

- I double checked for the `rp2040load` file/folder in the specified location:
    [image]
    [image]
    [image]

- This is the full verbose text:
    - [verbose_output.txt](https://github.com/arduino/ArduinoCore-mbed/files/6364608/verbose_output.txt)

facchinm, 2021-04-26:
Hi @santaimpersonator ,
it looks like your antivirus got rid of `rp2040load`; we got some other similar reports and we are working to whitelist the file.
The repo is here https://github.com/arduino/rp2040tools in case you want to make sure it's not a virus :wink: 
To fix the situation you can temporarily download the windows binary from [here](https://github.com/arduino/rp2040tools/releases/download/1.0.3/rp2040tools-1.0.3-windows_386.tar.bz2) and extract it in `C:\Users\<username>\Documents\ArduinoData\packages\arduino\tools\rp2040tools\1.0.2` folder

facchinm, 2021-04-26:
/cc @umbynos @cmaglie @rsora

closed 2021-04-26 by santaimpersonator