arduino/ArduinoCore-mbed #210: Raspberry pi pico unable to use 2nd or 3rd serial port (Serial1 or Serial2)

MNS26, 2021-04-28:
I was redirected here form platfromIO to open an issue about it.
( https://github.com/platformio/platform-raspberrypi/issues/3 )

I've been trying it read my gps module for a project and decided to use the pi pico for it, however even though it "sees" the connection (no check wire connection message) it receives no data.

i used the default "FullExampel" file from the TinyGPS library 
i removed SoftwareSerial and replaced it with Serial1 (RP pi pico "should be able to have 2 serial connections)

GPS lock is established (red led)
[image]

data from serial console 
[image]

facchinm, 2021-05-27:
Hi @MNS26 and @Gerriko ,
I just posted a PR https://github.com/arduino/ArduinoCore-mbed/pull/240 to ease the creation of additiona serial objects without touching the core.
By apply it the resulting sketch will look like
```
UART Serial2(8,9);

void setup() {
  Serial.begin(115200);
  Serial1.begin(115200);
  Serial2.begin(115200);

  delay(100);
  Serial.println("Lets start");
  // Just a quick initial test to make sure I am seeing text in both my serial terminals
  for (auto i = 0; i < 6; i++) {
    delay(3000);
    Serial1.println("Hello world 1");  
    Serial2.println("Hello world 2");
  }
  Serial.println("We are finished first test");
}

void loop() {
  if (Serial2.available()) {      // If anything comes in Serial (USB),
    Serial1.write(Serial2.read());   // read it and send it out Serial1 (pins 0 & 1)
    Serial.print(".");          // output via usb just in case we don't get anything
  }

  if (Serial1.available()) {     // If anything comes in Serial1 (pins 0 & 1)
    Serial2.write(Serial1.read());   // read it and send it out Serial (USB)
    Serial.print(".");          // output via usb just in case we don't get anything
  }
}
```
I tested it on a Pico and couldn't reproduce the crash that @Gerriko is reporting. Could you test if the issue is reproducible after the PR is applied? Thx

MNS26, 2021-05-29:
> Hi @MNS26 and @Gerriko ,
> I just posted a PR #240 to ease the creation of additiona serial objects without touching the core.
> By apply it the resulting sketch will look like
>
> ```
> UART Serial2(8,9); [...]

@facchinm after i edited the pins_arduino.h file and tried to use your example i cant get it to compile, everytime i try to do it i get this

```
> Executing task: C:\Users\MS26\.platformio\penv\Scripts\platformio.exe run --environment pico helicopter <

Processing pico helicopter (platform: https://github.com/platformio/platform-raspberrypi.git; framework: arduino; board: pico)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.0.0+sha.095da74) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, raspberrypi-swd)
PACKAGES:
 - framework-arduino-mbed 2.1.0
 - tool-rp2040tools 1.0.2
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Wire>
|-- <TinyGPSPlus> 1.0.2
Building in release mode
Compiling .pio\build\pico helicopter\src\main.cpp.o
Compiling .pio\build\pico helicopter\src\main1.cpp.o
Compiling .pio\build\pico helicopter\lib1b0\Wire\Wire.cpp.o
Compiling .pio\build\pico helicopter\libf0e\TinyGPSPlus\TinyGPS++.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduinoVariant\double_tap_usb_boot.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\Interrupts.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\Serial.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\USB\PluggableUSBDevice.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\USB\USBCDC.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\USB\USBSerial.cpp.o
In file included from C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/platform_defs.h:12,
                 from C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/pico_platform/include/pico/platform.h:12,
                 from C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_base/include/pico.h:19,
                 from C:\Users\MS26\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO\double_tap_usb_boot.cpp:2:
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_timer/include/hardware/timer.h: In function 'uint32_t time_us_32()':       
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/regs/addressmap.h:56:20: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
   56 | #define TIMER_BASE 0x40054000
      |                    ^~~~~~~~~~
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_structs/include/hardware/structs/timer.h:33:38: note: in expansion of macro 'TIMER_BASE'
   33 | #define timer_hw ((timer_hw_t *const)TIMER_BASE)
      |                                      ^~~~~~~~~~
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_timer/include/hardware/timer.h:66:12: note: in expansion of macro 'timer_hw'
   66 |     return timer_hw->timerawl;
      |            ^~~~~~~~
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_timer/include/hardware/timer.h: In function 'bool time_reached(absolute_time_t)':
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/regs/addressmap.h:56:20: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
   56 | #define TIMER_BASE 0x40054000
      |                    ^~~~~~~~~~
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_structs/include/hardware/structs/timer.h:33:38: note: in expansion of macro 'TIMER_BASE'
   33 | #define timer_hw ((timer_hw_t *const)TIMER_BASE)
      |                                      ^~~~~~~~~~
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_timer/include/hardware/timer.h:110:19: note: in expansion of macro 'timer_hw'
  110 |     uint32_t hi = timer_hw->timerawh;
      |                   ^~~~~~~~
[... cut ...]
void loop() {
  if (Serial2.available()) {      // If anything comes in Serial (USB),
    Serial.write(Serial2.read());   // read it and send it out Serial1 (pins 0 & 1)
    Serial.println();          // output via usb just in case we don't get anything
  }
/*
  if (Serial1.available()) {     // If anything comes in Serial1 (pins 0 & 1)
    Serial2.write(Serial1.read());   // read it and send it out Serial (USB)
    Serial.print(".");          // output via usb just in case we don't get anything
  }*/
}
```

Gerriko, 2021-05-30:
@MNS26 you say " i edited the pins_arduino.h file "

When looking through the error log it tells you  warning: "PIN_SERIAL_TX" redefined" and "warning: "PIN_SERIAL_RX" redefined"

That suggests to me that Serial2 and Serial1 and trying to use the same PIN_SERIAL_TX and PIN_SERIAL_RX.

Maybe check your pins_arduino.h file again.

MNS26, 2021-05-31:
@Gerriko I have pin GP4 and pin GP5 for serial 1 (GPS) redefined and plan on using GP12 and GP13 for serial 0 (RC receiver).
In the pins_arduino.h file 
pin definition 
```
// Serial
#define PIN_SERIAL_TX (0ul)
#define PIN_SERIAL_RX (1ul)

#define PIN_SERIAL_TX2 (8ul)
#define PIN_SERIAL_RX2 (9ul)
```
serialX definition 
```
//SERIAL Setup
#define SERIAL_HOWMANY		2
#define SERIAL1_TX			(digitalPinToPinName(PIN_SERIAL_TX))
#define SERIAL1_RX			(digitalPinToPinName(PIN_SERIAL_RX))

#define SERIAL2_TX			(digitalPinToPinName(PIN_SERIAL_TX2))
#define SERIAL2_RX			(digitalPinToPinName(PIN_SERIAL_RX2))
```
i used GP0 and GP1 for a I2C sensor (mpu6050 with auxiliary I2C enabled for compass)

MNS26, 2021-06-11:
Is there anything else i can try to get it to work?
@facchinm do you have any idea why it does compile (but not work) when removing the `UART Serial2(4,5);` line?
Ive been trying things out in my free time here and there but i either fail to compile it (same thing as previous message) or the serial port is not made.

Should i see this project as a lost cause and "drop it"? or is there still a light _somewhere_ at the end of the tunnel.

resolvethis, 2021-06-16:
The proposed code did not compile as it was shown there.

`    UART Serial2(8, 9);  // did not compile`

I got the following error:

```
no matching function for call to 'arduino::UART::UART(int, int)'
...
note: candidate: arduino::UART::UART(int, int, int, int)
```

What worked for me was:

`    UART Serial2(8, 9, 0, 0);`

Full code:

```
    #include <Arduino.h>
    
    UART Serial2(8, 9, 0, 0);
    
    void setup() {
      Serial2.begin(9600);
      // ...
      Serial2.write("hello world");
    }
    
    void loop {}
```

MNS26, 2021-06-27:
@resolvethis Did you also edit the pins_arduino.h file? When i try and use your code (and adding the () to loop) i still get a error.

```
Processing pico helicopter (platform: https://github.com/platformio/platform-raspberrypi.git; framework: arduino; board: pico)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.1.0+sha.3c6cc84) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, raspberrypi-swd)
PACKAGES:
 - framework-arduino-mbed 2.1.0
 - tool-rp2040tools 1.0.2
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TinyGPSPlus> 1.0.2
Building in release mode
Compiling .pio\build\pico helicopter\src\main.cpp.o
Compiling .pio\build\pico helicopter\libf0e\TinyGPSPlus\TinyGPS++.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduinoVariant\double_tap_usb_boot.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\Interrupts.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\Serial.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\USB\PluggableUSBDevice.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\USB\USBCDC.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\USB\USBSerial.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\main.cpp.o
Compiling .pio\build\pico helicopter\FrameworkArduino\pinToIndex.cpp.o
In file included from C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/platform_defs.h:12,
                 from C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/pico_platform/include/pico/platform.h:12,
                 from C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_base/include/pico.h:19,
                 from C:\Users\MS26\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO\double_tap_usb_boot.cpp:2:
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_timer/include/hardware/timer.h: In function 'uint32_t time_us_32()':
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/regs/addressmap.h:56:20: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
   56 | #define TIMER_BASE 0x40054000
      |                    ^~~~~~~~~~
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_structs/include/hardware/structs/timer.h:33:38: note: in expansion of macro 'TIMER_BASE'
   33 | #define timer_hw ((timer_hw_t *const)TIMER_BASE)
      |                                      ^~~~~~~~~~
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_timer/include/hardware/timer.h:66:12: note: in expansion of macro 'timer_hw'
   66 |     return timer_hw->timerawl;
      |            ^~~~~~~~
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_timer/include/hardware/timer.h: In function 'bool time_reached(absolute_time_t)':
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/regs/addressmap.h:56:20: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
   56 | #define TIMER_BASE 0x40054000
      |                    ^~~~~~~~~~
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_structs/include/hardware/structs/timer.h:33:38: note: in expansion of macro 'TIMER_BASE'
   33 | #define timer_hw ((timer_hw_t *const)TIMER_BASE)
      |                                      ^~~~~~~~~~
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_timer/include/hardware/timer.h:110:19: note: in expansion of macro 'timer_hw'
  110 |     uint32_t hi = timer_hw->timerawh;
      |                   ^~~~~~~~
C:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/regs/addressmap.h:56:20: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
   56 | #define TIMER_BASE 0x40054000
      |                    ^~~~~~~~~~
[... cut ...]
*** [.pio\build\pico helicopter\firmware.elf] Error 1
==================================================================== [FAILED] Took 13.01 seconds ====================================================================
The terminal process "C:\Users\MS26\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'pico helicopter'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

but when i set ``#define SERIAL_HOWMANY`` to 1 it compiles without a problem.

MNS26, 2021-07-31:
@facchinm has there been any news on this issue? I have yet to find a way to use multiple serial ports at once.

aord, 2022-05-15:
Many thanks to those above, especially facchinm:
Below is what I used for my NEO GPS connected to the Pico.
I had trouble with Serial1, every time I used Serial1 along with Serial it would lock up the PC comport. I'm guessing that serial1 is the same as serial? I don't know, but the following code worked for me:

#include <TinyGPSPlus.h>

TinyGPSPlus gps;

// The serial connection to the GPS device
UART Serial2(8,9); 
//THE GPS TX goes to GP9 to work, 9 on the PICO to the recieve pin
//THE GPS TX goes to GP9 to work, 9 on the PICO to the recieve pin

void setup()
{
  Serial.begin(9600);
  Serial2.begin(9600); //You might be 4800 depending on your GPS?
}

void loop()
{
  
  delay(500);
  // This sketch displays information every time a new sentence is correctly encoded.
  while (Serial2.available() > 0)
  {
    if (gps.encode(Serial2.read()))
      displayInfo();
  }
    
  if (millis() > 5000 && gps.charsProcessed() < 10)
  {
    Serial.println(F("No GPS detected: check wiring."));
    //while(true);
    delay(7000);
    Serial.println("Waited 7 sec for you to wire");
  }
}

void displayInfo()
{
  Serial.print(F("Location: ")); 
  if (gps.location.isValid())
  {
    Serial.print(gps.location.lat(), 6);
    Serial.print(F(","));
    Serial.print(gps.location.lng(), 6);
  }
  else
  {
    Serial.print(F("INVALID"));
  }

  Serial.print(F("  Date/Time: "));
  if (gps.date.isValid())
  {
    Serial.print(gps.date.month());
    Serial.print(F("/"));
    Serial.print(gps.date.day());
    Serial.print(F("/"));
    Serial.print(gps.date.year());
  }
  else
  {
    Serial.print(F("INVALID"));
  }

  Serial.print(F(" "));
  if (gps.time.isValid())
  {
    if (gps.time.hour() < 10) Serial.print(F("0"));
    Serial.print(gps.time.hour());
    Serial.print(F(":"));
    if (gps.time.minute() < 10) Serial.print(F("0"));
    Serial.print(gps.time.minute());
    Serial.print(F(":"));
    if (gps.time.second() < 10) Serial.print(F("0"));
    Serial.print(gps.time.second());
    Serial.print(F("."));
    if (gps.time.centisecond() < 10) Serial.print(F("0"));
    Serial.print(gps.time.centisecond());
  }
  else
  {
    Serial.print(F("INVALID"));
  }

  Serial.println();
}