arduino/ArduinoCore-mbed #211: STM32LO LoRa Murata Module I2C Communication with PortentaH7 M7 core

hpssjellis, 2021-04-29:
Does anyone know if the STM32LO Module I2C  pins ```PB_9``` SDA,  ```PB_8``` SCL are able to be used?

Using this pdf on page 7  [https://content.arduino.cc/assets/SL-ASX00026.PDF](https://content.arduino.cc/assets/SL-ASX00026.PDF)

[image]

Looks easy enough to communicate with the STM32LO  Murata module , I can load simple Slave or Master wire.h programs onto it, but when I try to read it using the PortentaH7 M7 core also running wire.h slave or master I get nothing. Any suggestions?

P.S. Communicating with the STM32LO Lora module is a bit complex @facchinm has a method [here](https://github.com/arduino-libraries/MKRWAN/issues/87#issuecomment-826180994)

Also does anyone know if SPI is working?

facchinm, 2021-04-29:
Hi Jeremy,
if you look at the top page, only the UART and a couple of control GPIOs are actually routed 
[image]
So, no way to communicate via other means :neutral_face:

closed 2021-04-29 by facchinm

hpssjellis, 2021-04-29:
Thanks @facchinm that explains why I can only communicate from the Portenta  M7 Core to the Murata LoRa Module  using: 

```
UART mySerial3(PJ_8,  PJ_9,  NC, NC); //TX, TR, RTS, CTS  NOTE: NC means not connected

```

It works fine, so I should be good.