arduino/ArduinoCore-mbed #212: Unable to get I2C working on Raspberry Pi Pico

fsievers, 2021-05-01:
No matter what I try, I can't get I2C to work. I'm using the following i2c-scan to discover a SSD1306 display:
```
#include <Arduino.h>
#include <Wire.h>

void setup()
{
  Wire.begin();

  Serial.begin(9600);
  while (!Serial)
    ; // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner");
}

void loop()
{
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");
  
  nDevices = 0;
  for (address = 1; address < 127; address++)
  {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address < 16)
        Serial.print("0");
      Serial.print(address, HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error == 4)
    {
      Serial.print("Unknown error at address 0x");
      if (address < 16)
        Serial.print("0");
      Serial.println(address, HEX);
    }
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000); // wait 5 seconds for next scan
}
```
But I always get `No I2C devices found`. In `pins_arduino.h` it says, that SDA is on GPIO 6 and SCL is on GPIO 7 (see https://datasheets.raspberrypi.org/pico/Pico-R3-A4-Pinout.pdf). But the program is not able to discover the OLED display.

In a second try, I added
```
#define PIN_WIRE_SDA (8u)
#define PIN_WIRE_SCL (9u)
```
after the `Wire.h` include and tried to get it working on the Raspberry Pi Pico default I2C0 connections. But this doesn't work, too.

But I can confirm, that the Pico and OLED are OK and I2C is working, if I use https://github.com/earlephilhower/arduino-pico installed via Board Manager. The project uses GPIO 4 and 5 for I2C0 (Wire) by default. And I can even change the GPIOs via
```
Wire.setSDA(8);
Wire.setSCL(9);
```
in the `setup()` function right before doing `Wire.begin()`.

What can I do, to get I2C working with the ArduinoCore-mbed?

facchinm, 2021-05-04:
Hi @fsievers ,
about the I2C scan, there's a know issue with `Wire.endTransmission()` for zero bytes length packets (like the ones used during scanning).
However, I tested an SSD1306 oled display with the Pico and it works just fine, when connected to pin GP6 and GP7, even if the scan fails.
The issue with different pinmuxing is being discussed here https://github.com/arduino/ArduinoCore-mbed/issues/194 , but summarising:
* you can't change the pins by adding definitions inside the sketch (since it's a different compilation unit)
* we are not going to implement `setSCL` and `setSDA` in this core (since they are not part of the official APIs)
* the preferred way to change the pinmux is by creating a new object and then use it.

For example, using the standard Adafruit library, you can write 
```
MbedI2C myi2c(p14,p15);
Adafruit_SSD1306 display = Adafruit_SSD1306(128, 32, &myi2c);
```
and the relevant I2C will be allocated on pins 14/15

closed 2021-05-04 by facchinm

fsievers, 2021-05-04:
Ok, I've checked it with the ssd1306 I2C example and the display works for all I2C0 pins (sorry I didn't checked the display itself without scan). But on the I2C1 pins, the display just give me the first frame and after that it doesn't show any further frame.

But for this topic, I think it's solved, regarding to I2C scan problem and known issue of `Wire.endTransmission()`.

facchinm, 2021-05-04:
I think you only see the first image since the example defaults to 
```
  #define BUTTON_A  9
  #define BUTTON_B  6
  #define BUTTON_C  5
```
so, as soon as BUTTON_B gets configured, the I2C functionality is lost :slightly_smiling_face: 
I'll keep you posted as soon as the scan is fixed :wink: