arduino/ArduinoCore-mbed #215: 2.0.0 does not enumerate the Pico **Port** stand alone.

zfields, 2021-05-05:
Arduino IDE 1.8.13
Ubuntu Linux 20.10

I purchased a Raspberry Pi Pico board.
Added Arduino MBed OS RP2040 Boards v2.0.0

The Pico would not show up under **Tools > Port**

I tried restarting the IDE, then I installed all updates and restarted the machine, nothing worked.

So, I added the Additional Boards Manager URL for **Raspberry Pi Pico/RP2040** board by Earle F. Philhower and installed the board, then both Board definitions (official and unofficial) could enumerate the port.

 I believe the official Board Support is missing the component that enumerates the board for Linux.

facchinm, 2021-05-06:
Hi @zfields ,
the first time you need to upload "blindly" with the Pico in bootloader mode (the port doesn't appear in the menu but the upload succeeds anyway). After that, the port will appear and you can upload in the usual way.
Let me know if it works :wink:

closed 2021-05-06 by facchinm

chowha, 2021-05-14:
Hi I created a game pad using the Mbed USBHID class. Doing this removes the serial port so I have to go into bootloader mode again for every upload and I can't read or write to the serial port. Is there a solution to this?

facchinm, 2021-05-14:
@chowha yup, just include "PluggableUSBHID.h" instead than "USBHID.h" and it will expose both the serial port and the HID device :slightly_smiling_face:

chowha, 2021-05-15:
Thanks Martino, that's works perfectly. You are the man!