arduino/ArduinoCore-mbed #220: Pico onboard temperature sensor

benevpi, 2021-05-12:
As far as  I can see, there's no ability to access the onboard temperature sensor on RP2040. It should be on ADC 4, but only 0-3 are available.

jessestr, 2021-05-19:
Try the unofficial Arduino port by Earle: https://github.com/earlephilhower/arduino-pico

There's a method called analogReadTemp();

Gerriko, 2021-05-22:
I just used the Pico C SDK code, as the pico-SDK is included with the official Pico board manager files and it works in the Arduino IDE - as in it compiled and prints out values.

```
// 12-bit conversion, assume max value == ADC_VREF == 3.3 V
const float CONVERSION_FACTOR = 3.3f / (1 << 12);

void setup() {
  // put your setup code here, to run once:
  adc_init();
  adc_set_temp_sensor_enabled (true);
  // Select ADC input 4 for internal temperature sensor
  adc_select_input(4);

}

void loop() {
  // put your main code here, to run repeatedly:
  
  uint16_t adc = adc_read();
  float ADC_Voltage = float(adc) * CONVERSION_FACTOR;
  float T = 27 - (ADC_Voltage - 0.706)/0.001721;        // formula found on page 71 (section 4.1.1. hardware_adc) of the Raspberry Pi Pico C/C++ SDK documentation
  
  Serial.println("Raw value: 0x" + String(adc, HEX) + " | ADC voltage: " + String(ADC_Voltage, 2) + "V | Temp: " + String(T, 2) + "C");
  
  delay(1000);
}

```

benevpi, 2021-05-25:
Thanks @Gerriko that solves the problem perfectly.

closed 2021-05-25 by benevpi

DeimosHall, 2023-07-16:
> I just used the Pico C SDK code, as the pico-SDK is included with the official Pico board manager files and it works in the Arduino IDE - as in it compiled and prints out values.
>
> ```
> // 12-bit [...]

I created a library based on your code to make it simplier, I hope you can find it useful: https://deimoshall.dev/blog/software/iot/how-to-use-onboard-raspberry-pi-pico-temperature-sensor/
The code example is:
```
/*
 * Test sketch for the RP2040 CPU temperature sensor library
 * Author: Francisco Torres (deimoshall.dev)
 * Date: July 2023
 * License: MIT
 * Repository: https://github.com/DeimosHall/RP2040_CPU_Temperature.git
*/

#include <CPU.h>  // Include the library to use the CPU temperature sensor

CPU cpu;  // Create an instance of the CPU temperature sensor

void setup() {
  Serial.begin(9600);
  cpu.begin();  // Initialize the CPU temperature sensor
}

void loop() {
  // Print the CPU temperature in Celsius
  Serial.println("CPU temperature: " + String(cpu.getTemperature()) + " °C");
  delay(1000);
}
```