arduino/ArduinoCore-mbed #221: F_CPU define is missing

blackketter, 2021-05-13:
The preprocessor define for CPU speed, F_CPU, is not defined for RaspberryPi Pico, but seems to be defined for other Arduino boards.  Any sketch or library that uses this definition will fail to build.

Thanks!

RudolfAtHome, 2021-06-03:
That would be a very important addition! The only library for the DHT22 sensor therefore does not work. And some others probably too.

RudolfAtHome, 2021-06-03:
Any fast work around?

facchinm, 2021-06-07:
A library using F_CPU to calculate timing is unlikely to work with this core (since the underlying RTOS does not guarantee that a busy loop is not getting interrupted).
For the DHT22, Adafruit's DHT library has already been ported to a plethora of architectures and just works fine without `F_CPU` definition. For any other library, I'd add something like https://github.com/facchinm/ArduinoCore-mbed/pull/new/f_cpu_defined to make the compiler happy but the user aware that the resulting sketch will likely not work.
@sebromero any idea / comment?

blackketter, 2021-06-07:
My use case was for printing debug information, so setting  F_CPU to zero isn't that helpful.  The fact that some libraries use the CPU speed for timing without taking architecture into account need to be fixed in any case.