arduino/ArduinoCore-mbed #223: Can't program via Arduino IDE the Pi Pico

WanaGo, 2021-05-19:
Hi,
Just having problems getting even a blink sketch programmed into the Pi Pico from the Arduino IDE.
Just updated to 2.1.0 of this, but still no joy.
Pushing the BOOTSEL and plugging in USB, no com port comes up which I think is expected. Plugging in normally it comes up as COM10, so that seems fine. RPI-RP2 comes up in device manager under Portable Devices in BOOTSEL mode.

Using Thonny, it detects the board and it prompts to update the firmware, so did that, but that didnt change anything. Closing Thonny and going back to the Arduino IDE, I still cannot program in BOOTSEL mode or normal mode. I can load programs fine in Thonny.

Closing Thonny completly, opening the Arduino IDE and trying both in BOOTSEL and normal COM10 selected, I get the following:

`Sketch uses 78640 bytes (0%) of program storage space. Maximum is 16777216 bytes.
Global variables use 52656 bytes (19%) of dynamic memory, leaving 217680 bytes for local variables. Maximum is 270336 bytes.
C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2/rp2040load -v -D C:\Users\xxx\AppData\Local\Temp\arduino_build_233391/Blink.ino.elf 
rp2040load 1.0.1 - compiled with go1.15.8
.....................
An error occurred while uploading the sketch
`

`Sketch uses 78640 bytes (0%) of program storage space. Maximum is 16777216 bytes.
Global variables use 52656 bytes (19%) of dynamic memory, leaving 217680 bytes for local variables. Maximum is 270336 bytes.
Forcing reset using 1200bps open/close on port COM10
C:\Users\james\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2/rp2040load -v -D C:\Users\james\AppData\Local\Temp\arduino_build_233391/Blink.ino.elf 
rp2040load 1.0.1 - compiled with go1.15.8
.....................
An error occurred while uploading the sketch
`

I just cant figure out what I am missing here.

[image]

Thanks

WanaGo, 2021-05-19:
Managed to sort it after trying a fresh install on my laptop which has not had Arduino IDE on it, and that all worked perfectly. Back on my desktop, is what I got above.

Went into device manager and with the Pi Pico attached to the USB, right clicking on each of the Pi items and uninstalling the driver, and then doing the same under BOOTSEL, I then ran the post_install.bat from the installer, and plugged the pi back in again, and now it all seems to work as it should.

`Sketch uses 78640 bytes (0%) of program storage space. Maximum is 16777216 bytes.
Global variables use 52656 bytes (19%) of dynamic memory, leaving 217680 bytes for local variables. Maximum is 270336 bytes.
C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2/rp2040load -v -D C:\Users\xxx\AppData\Local\Temp\arduino_build_356839/Blink.ino.elf 
rp2040load 1.0.1 - compiled with go1.15.8
Loading into Flash: [==============================]  100%

`

sebromero, 2021-05-19:
@WanaGo Glad to hear you sorted it out. Can we close this issue?

closed 2021-05-19 by WanaGo