arduino/ArduinoCore-mbed #224: Error while using Servo.h

abhisheksaraiya, 2021-05-19:
Using the standard sweep example and get the following error.
Using with pi pico. 

```
WARNING: library Servo claims to run on avr, megaavr, sam, samd, nrf52, stm32f4, mbed architecture(s) and may be incompatible with your current board which runs on mbed_rp2040 architecture(s).
libraries\Servo\mbed\Servo.cpp.o: In function `Servo::attach(int, int, int)':
C:\Program Files (x86)\Arduino\libraries\Servo\src\mbed/Servo.cpp:65: undefined reference to `digitalPinToPinName(unsigned char)'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Raspberry Pi Pico.
```
Have already selected the mbed rp2040 board in the Arduino IDE:

facchinm, 2021-05-19:
Hi @abhisheksaraiya ,
should be fixed by https://github.com/arduino-libraries/Servo/commit/2298fb27b9e60b096b5d80a267bf926a6c6f31f7 ; you you give it a try? Thanks

abhisheksaraiya, 2021-05-20:
Awesome, compiles fine. Thanks.

closed 2021-05-20 by abhisheksaraiya