arduino/ArduinoCore-mbed #228: Compilation errors: libmbed.a libcc_310_core.a libcc_310_ext.a libcc_310_trng.a not found

carlosperate, 2021-05-20:
Arduino IDE version: 1.8.12 (v2-beta also has issues compiling, although the output is not printed).
Board: Arduino Nano 33 BLE Sense
Arduino Core: Arduino Mbed OS Nano Boards 2.1.0
Additional library: Arduino_CMSIS-DSP https://github.com/arduino-libraries/Arduino_CMSIS-DSP

This compilation error can be easily reproduce by adding `#include "arm_math.h"` to the blinky demo:
```
arm-none-eabi-g++: error: ../../../../../../Users/carlos/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.1.0/variants/ARDUINO_NANO33BLE/libs/libmbed.a: No such file or directory
arm-none-eabi-g++: error: ../../../../../../Users/carlos/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.1.0/variants/ARDUINO_NANO33BLE/libs/libcc_310_core.a: No such file or directory
arm-none-eabi-g++: error: ../../../../../../Users/carlos/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.1.0/variants/ARDUINO_NANO33BLE/libs/libcc_310_ext.a: No such file or directory
arm-none-eabi-g++: error: ../../../../../../Users/carlos/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.1.0/variants/ARDUINO_NANO33BLE/libs/libcc_310_trng.a: No such file or directory
exit status 1
Error compiling for board Arduino Nano 33 BLE.
```

Originally I encountered this issue when trying to build this TensorFlow example: https://github.com/googlecreativelab/tf4micro-motion-kit
I reported the issue here: https://github.com/googlecreativelab/tf4micro-motion-kit/issues/3
I could not find a direct include of `arm_math.h` in that source code, maybe one of the dependencies includes it, or maybe it can be triggered by other means.

carlosperate, 2021-05-20:
Is it possible that the multiple `../` are not navigating the directory correctly, looks like the files are there

```
$ l /Users/carlos/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.1.0/variants/ARDUINO_NANO33BLE/libs/
total 12536
drwxr-xr-x   6 carlos  staff   192B 17 May 09:54 .
drwxr-xr-x  14 carlos  staff   448B 17 May 09:54 ..
-rw-r--r--   1 carlos  staff   222K 17 May 09:51 libcc_310_core.a
-rw-r--r--   1 carlos  staff    12K 17 May 09:51 libcc_310_ext.a
-rw-r--r--   1 carlos  staff   2.8K 17 May 09:51 libcc_310_trng.a
-rw-r--r--   1 carlos  staff   5.4M 17 May 09:55 libmbed.a
```

facchinm, 2021-05-21:
@carlosperate thanks for reporting! @sebromero could you test on your mac to confirm the bug?

carlosperate, 2021-05-21:
Just as another datapoint I've also tried v2.0.0 of the "Arduino Mbed OS Nano Boards" core and having exactly the same issue, and I can also confirm that the files are "there" (although still not sure how that relative path resolves):

```
arm-none-eabi-g++: error: ../../../../../../Users/carlos/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.0.0/variants/ARDUINO_NANO33BLE/libs/libmbed.a: No such file or directory
arm-none-eabi-g++: error: ../../../../../../Users/carlos/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.0.0/variants/ARDUINO_NANO33BLE/libs/libcc_310_core.a: No such file or directory
arm-none-eabi-g++: error: ../../../../../../Users/carlos/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.0.0/variants/ARDUINO_NANO33BLE/libs/libcc_310_ext.a: No such file or directory
arm-none-eabi-g++: error: ../../../../../../Users/carlos/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.0.0/variants/ARDUINO_NANO33BLE/libs/libcc_310_trng.a: No such file or directory
exit status 1
Error compiling for board Arduino Nano 33 BLE.
```

```
 $ l /Users/carlos/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.0.0/variants/ARDUINO_NANO33BLE/libs/
total 12808
drwxr-xr-x   6 carlos  staff   192B  8 Apr 17:31 .
drwxr-xr-x  14 carlos  staff   448B  8 Apr 17:31 ..
-rw-r--r--   1 carlos  staff   222K  8 Apr 17:27 libcc_310_core.a
-rw-r--r--   1 carlos  staff    12K  8 Apr 17:27 libcc_310_ext.a
-rw-r--r--   1 carlos  staff   2.8K  8 Apr 17:27 libcc_310_trng.a
-rw-r--r--   1 carlos  staff   5.4M  8 Apr 17:32 libmbed.a
```

sebromero, 2021-05-21:
I can't reproduce it. I have the same setup except of the IDE, I'm on 1.8.13. It works for me when compiling for the Nano 33 BLE. It does not work however for the Nano RP2040 Connect as the mcu is not supported by the library:
```/Users/sebastianhunkeler/Documents/Arduino/libraries/Arduino_CMSIS-DSP/src/Source/TransformFunctions/arm_bitreversal2.S:145: Error: selected processor does not support `it LS' in Thumb mode```

sebromero, 2021-05-21:
@carlosperate I recommend to uninstall the library, uninstall the core, upgrade to the newest version of the Arduino IDE, install the core and the library again and try compiling your sketch again.

carlosperate, 2021-05-21:
Thanks @sebromero I've upgraded to the IDE v1.8.15 and it worked without having to reinstall the cores or libraries.
Looks like it was an already resolved problem with the IDE after all, thanks everyone!

closed 2021-05-21 by carlosperate