arduino/ArduinoCore-mbed #232: USBMIDI::write() compiles but doesn't work on Raspberry Pi Pico

skybrian, 2021-05-23:
Raspberry Pi Pico board support includes the USBMIDI.h header from MBed OS, but it doesn't seem to work. When USBMIDI::write is called it prints a garbage character to the serial console and crashes.

I expected it either to work or result in some kind of compile error.

Sample code attached.

[pressure_sensor_test.ino.txt](https://github.com/arduino/ArduinoCore-mbed/files/6527612/pressure_sensor_test.ino.txt)

facchinm, 2021-06-07:
Hi @skybrian ,
the default USB* libraries from mbed will replace the CDC Serial port with their device, so they are not suitable for an Arduino-like usage.
If you take a look in `libraries` folder inside the core, a bunch of them have been ported to the Pluggable infrastructure; we still didn't port USBMIDI but a user did it https://github.com/tttapa/Arduino-mbed-PluggableUSBMIDI-experiment .
If you want to test and report your results it would be great!

skybrian, 2021-06-07:
I switched to earlephilhower/arduino-pico which works for me, so I’m not working on this anymore.

Maybe there is some way to remove the footgun until this feature is finished? Perhaps add a comment about what it does?

rltvty, 2022-01-21:
@skybrian I'm running into similar issues getting MIDI to work with an adafruit feather RP2040.  Could you provide some details as to how you got `USBMIDI::write` working with `earlephilhower/arduino-pico` ?

Thanks in advance!

rltvty, 2022-01-21:
nevermind, I found an example: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/MIDI/midi_test/midi_test.ino