arduino/ArduinoCore-mbed #233: Pico RP2040 - Debugging with Segger J-Link

rei-vilo, 2021-05-23:
How to debug an Arduino program with the Segger J-Link? 

- Core: Arduino Mbed OS RP2040 Boards
- Board: Pico RP2040 board
- Option: `-g3 -ggdb`
- Programmer: Segger J-Link

The mbed RTOS plug-in doesn't seem to be listed among the standard options for the JLinkGDBServer. 

The debugger stops at the two breakpoints...

```
Breakpoint 1, loop () at embed1.ino:94
94	    blink(myLED, 4, 333);

Breakpoint 2, blink (pin=25 '\031', times=times@entry=4 '\004', ms=ms@entry=333, level=level@entry=true) at LocalLibrary.cpp:27
27	        digitalWrite(pin, (level ^ (i % 2)) ? HIGH : LOW);
```

...then loops endlessly. Pausing shows the error...

```
Program received signal SIGTRAP, Trace/breakpoint trap.
0x1000636a in mbed_die ()
```
 
...with code 

```
0x10006364: 10 b5           	push	{r4, lr}
0x10006366: 00 f0 55 f8     	bl	0x10006414 <core_util_critical_section_enter>
0x1000636a: fe e7           	b.n	0x1000636a <mbed_die+6>
```

Thank you!

facchinm, 2021-06-07:
Hi @rei-vilo ,
unfortunately `RTX5` (the rtos used as base for mbed) is not supported by many debuggers (the only official one is https://github.com/pyocd/pyOCD ). I started to port the rtos awareness to openocd some time ago but it's still a work in progress (https://github.com/facchinm/OpenOCD/tree/rtx5).
The situation, as far as I know, is:
* pyocd : supports rtx5 + rp2040 but the only compatible debugger is `picoprobe` 
* openocd: supports rp2040 + jlink (in 0.11-arduinox builds) but no rtos awareness
* jlink: supports rp2040 but no rtos
It might change in the near future since we are in touch with Segger but I can't promise anything. If you are interested in any openocd specific build I'll be glad to send you more instructions :wink:

rei-vilo, 2021-06-08:
Thank you for the detailed answer and multiple pointers. 

It seems rather surprising that two industry standards, ARM mbed and Segger, haven't teamed up to bring J-Link to mbedOS. 

As a matter of facts, I've already asked Segger a related question at [Raspberry Pi Pico RP2040 - Cortex-M HardFault Exception](https://forum.segger.com/index.php/Thread/7977-Raspberry-Pi-Pico-RP2040-Cortex-M-HardFault-Exception/).

Anyway, I'll go through the different options. I have a second Pico as `picoprobe` and this solution seems to be supported by the cortex-debug extension on Visual Studio Code.