arduino/ArduinoCore-mbed #234: Confusing options for Mbed cores

rei-vilo, 2021-05-23:
To which core this repository belongs to?

[image]

Thank you!

per1234, 2021-05-23:
All of them. The split into "families" was implemented in the 2.0.0 release:
https://github.com/arduino/ArduinoCore-mbed/releases/tag/2.0.0
>Improve packaging scripts by splitting the core into families [1]
>...
>[1] Breaking changes:
>Following a broad community request, we targeted two pain point of existing core:
>
>- installation speed
>- compilation speed
>They are both very noticeable on Windows platforms, where file access is much slower and the antivirus can potentially check every file being compiled.
>
>To fix the first, we removed debugging information from libmbed.a (to make the final library much smaller) and we split the core into different packages. This only happens at release time, if you are developing nothing changes.

Here's the script that does it:
https://github.com/arduino/ArduinoCore-mbed/blob/master/package.sh
and the one for the deprecated mono-platform:
https://github.com/arduino/ArduinoCore-mbed/blob/master/package_full.sh

- https://github.com/arduino/ArduinoCore-mbed/blob/master/edge.variables
- https://github.com/arduino/ArduinoCore-mbed/blob/master/nano.variables
- https://github.com/arduino/ArduinoCore-mbed/blob/master/rp2040.variables
- https://github.com/arduino/ArduinoCore-mbed/blob/master/portenta.variables
- https://github.com/arduino/ArduinoCore-mbed/blob/master/full.variables

rei-vilo, 2021-05-24:
Thank you for the detailed explanations!

closed 2021-05-24 by rei-vilo