arduino/ArduinoCore-mbed #235: Ticker stops working after some time of use

beegee-tokyo, 2021-05-24:
I have an application where I attach/detach a ticker to control a second thread. Everything works fine for ~1 hour, then suddenly the ticker stops to work and the callback is no longer called. I need to use attach/detach because the tickers are used as watchdog while waiting for an external event. If the event happens, I detach the ticker, otherwise I wait for the callback to be triggered to detach the ticker and react the missing event. I looked as well into the Timeout class, but there is no function to stop the timeout. So if my external event happens, I cannot stop the timeout callback. 

Anyone has an idea? I created a small test application to confirm the behavior and it shows the same problem.
Example code:
```c++
#include <Arduino.h>

#include "mbed.h"
#include "rtos.h"
#include <cmsis_os.h>

using namespace std::chrono_literals;
using namespace std::chrono;
using namespace rtos;
using namespace mbed;

/** Timer for periodic sending */
mbed::Ticker test_ticker[10];
/** Semaphore to signal loop() a send request */
rtos::Semaphore sema0(1);

Thread events_thread_id;

/** Thread to handle the events */
void events_thread(void);

void cb_ticker0(void)
{
	sema0.release();
}

Timeout timeout;

void setup()
{
	// Initialize Serial for debug output
	Serial.begin(115200);

	time_t timeout = millis();
	while (!Serial)
	{
		delay(250);
		if ((millis() - timeout) > 5000)
		{
			break;
		}
	}

	Serial.println("---------------------------------------------");
	Serial.println("Ticker Test");
	Serial.println("---------------------------------------------");

	Serial.println("Taking the semaphore");
	// Take all semaphores
	sema0.acquire();

	Serial.println("Starting thread to handle the events");
	events_thread_id.start(events_thread);
	events_thread_id.set_priority(osPriorityAboveNormal);

	Serial.println("Starting the tickers");
	// Start the tickers
	test_ticker[0].attach(&cb_ticker0, std::chrono::duration<time_t>{1});
}

void loop()
{
	Serial.print("Loop awake at ");
	Serial.println(millis() / 1000);
	delay(10000);
}

void events_thread(void)
{
	while (true)
	{
		// Wait for event
		sema0.acquire();
		Serial.print("T0 triggered at ");
		Serial.println(millis() / 1000);
		test_ticker[0].detach();
		test_ticker[0].attach(&cb_ticker0, std::chrono::duration<time_t>{1});
		yield();
	}
}
```

Log output:
```log

00.000: ---------------------------------------------
Ticker Test
---------------------------------------------
Taking all semaphores
Starting thread to handle the events
Starting the tickers
Loop awake at 0
00.986: T0 triggered at 1
01.976: T0 triggered at 2
02.969: T0 triggered at 3
03.959: T0 triggered at 4
04.946: T0 triggered at 5
05.925: T0 triggered at 6
07.023: T0 triggered at 7
07.999: T0 triggered at 8
08.990: T0 triggered at 9
09.539: Loop awake at 10
09.967: T0 triggered at 10
10.960: T0 triggered at 11
11.936: T0 triggered at 12
13.028: T0 triggered at 13
14.013: T0 triggered at 14
15.001: T0 triggered at 15
15.995: T0 triggered at 16
16.970: T0 triggered at 17
17.957: T0 triggered at 18
18.948: T0 triggered at 19
19.167: Loop awake at 19
...
...
01:11:25.994: T0 triggered at 4286
01:11:26.980: T0 triggered at 4287
01:11:27.970: T0 triggered at 4288
01:11:28.977: T0 triggered at 4289
01:11:29.963: T0 triggered at 4290
01:11:30.938: T0 triggered at 4291
01:11:31.443: Loop awake at 4291
01:11:31.925: T0 triggered at 4292
01:11:32.933: T0 triggered at 4293
01:11:33.915: T0 triggered at 4294
01:11:40.684: Loop awake at 4301
01:11:50.304: Loop awake at 4310
01:11:59.966: Loop awake at 4320
01:12:09.544: Loop awake at 4330
01:12:19.120: Loop awake at 4339
01:12:28.690: Loop awake at 4349
01:12:38.360: Loop awake at 4358
01:12:47.890: Loop awake at 4368
01:12:57.482: Loop awake at 4378
```

beegee-tokyo, 2021-05-24:
Running a different test (without detach()/attach()), just let the ticker run, gives the same result. After ~4294 seconds the ticker stops working.

facchinm, 2021-06-07:
Hi @beegee-tokyo ,
thanks for reporting! Which board are you targeting?
The error really looks like an unsigned overrun (as if the next timeout was set to SOME_MAX + 1 which ends up being a very small number) so the issue might be in the mbed porting layer for that particular MCU.

beegee-tokyo, 2021-06-07:
@facchinm
target board is the Pico with the RP2040.
I found a workaround using low level functions of the RP2040 SDK.

facchinm, 2021-06-07:
Thanks for the info! The problem might be indeed in the porting layer since 4294 seconds are almost UINT32_MAX (since the timer ticks at 1MHz) https://github.com/arduino/mbed-os/blob/rp2040_pr/targets/TARGET_RASPBERRYPI/TARGET_RP2040/us_ticker.c .
I need to reread [the specifications](https://os.mbed.com/docs/mbed-os/v6.11/mbed-os-api-doxy/group__hal__us__ticker.html#ga46439f1db7a3f469c3bd4e32cabc01cb) to understand how to properly handle the rollover; in the meantime, thanks again!

beegee-tokyo, 2021-06-07:
@facchinm
Thanks for looking into it. I know Arduino Mbed OS support is just starting and I appreciate your support.

facchinm, 2021-07-09:
Should be fixed now since we merged https://github.com/arduino/ArduinoCore-mbed/commit/a09bee8418012267fde2d8f0bbd90fa7570a23ed#diff-eedd8d3e83f8acbff34b5f185f985581038e22debedaceaf1c09139175d3050c (a core release is incoming, but can be already tested using the core from git :wink: )

closed 2021-07-09 by facchinm

beegee-tokyo, 2021-07-09:
@facchinm 
Thank you, I will test over the weekend.