arduino/ArduinoCore-mbed #238: Feature Request: Dual Core Memory Flexibility

hpssjellis, 2021-05-26:
**Feature Request: Dual Core Memory Flexibility**

What makes a dual core Arduino so impressive is the ability to quickly exchange data between the cores. A lot of work has been done to make RPI requests successful, and we appreciate that RPI works well and is easy to understand. However in this [thread (passing-variable-from-cm7-to-cm4-using-rpc)](https://forum.arduino.cc/t/passing-variable-from-cm7-to-cm4-using-rpc/692026/11?u=jerteach) on the Arduino Forum we are noticing some speed issues with RPI that makes some dual core interactions too slow. 

Can some time be spent on checking if there is a faster way to exchange data between the cores, perhaps shared memory or even a bit more flexibility to assign different amounts of memory to different cores. I guess a start to this would be an explanation of how the memory is used on the Portenta. When I was working with the LoRa / Murata module it seemed to use memory that was outside of either core. Could some of this memory be used to shared a small amount of data between the cores?

@facchinm, I realize the answer is probably NO, but thought I would ask.

facchinm, 2021-05-27:
Hi Jeremy,
we tried to tackle the topic in this issue (https://github.com/arduino/ArduinoCore-mbed/issues/145) which resulted in this sample code (https://github.com/MatInGit/PortentaRPCMemory). 
The H747 is a bit of a strange beast since memory areas are accessed via different buses from M7 or M4. This can have any sort of unpredictable behavior is the data access is not guarded through the hardware semaphore (as wrapped in the RPC library for example). Moreover, data on M7 is usually cached, so it's not guaranteed to be written back to the shared memory even when it says so.
I agree that, right now, it's just suitable for "slow" communication (calling functions, not exchanging data).
On the LoRa / murata module topic, since the communication method between the processors is the serial port, none of these thing apply, and the RPC mechanism can happily decide its own pace.
@sebromero would it be a good idea to refactor the shared memory example, turn it into a library and provide a convenient API?

hpssjellis, 2021-05-27:
@facchinm Does that mean I could use both cores to do serial communication with the Murata memory? One core sending data and the other core reading the data. That would be easy for me to test using your version of the GrumpyOldPizza library.  

But would Serial communication be any faster than RPC? Presently I think I was having about a 20-40 ms lag time when communicating to the M7 core from the LoRa module. 

I don't think it will be any faster, but might just test it for fun.

hpssjellis, 2021-05-28:
After spending a day with the Murata memory, I got the M7 and M4 cores exchanging data at a rate of 3 ms per character. The order messed up at 2 ms per char, but it occurred to me that it was just serial communication, so I then made some code using TX and RX  and connected pins TX D14 and RX D13 and that was much faster sending accurate order of data even at 0.001 ms. I guess this is what  issue ([#145](https://github.com/arduino/ArduinoCore-mbed/issues/145)) is trying to do but using RPC
 ([https://github.com/MatInGit/PortentaRPCMemory](https://github.com/MatInGit/PortentaRPCMemory)).

My code is more basic and not well written but here it is.

```
// Easy Dual Core program setup

// Need to connect pin TX D14 and RX D13 together.

#include <Arduino.h>

//////////////////// Start All Core M7 Programing /////////////////////
#ifdef CORE_CM7 

UART myUART3(PA_9,  PA_10, NC, NC); //TX, TR, RTS, CTS  NOTE: NC means not connected

void setup() { 
   bootM4();    
   Serial.begin(115200);
   myUART3.begin(9600);   

}

void loop() {
  if (myUART3.available()) {          // If anything comes in Serial3 
     Serial.write(myUART3.read());   // Read it and send it out Serial (USB)
  }

}

#endif

//////////////////// End All Core M7 Programing /////////////////////

//////////////////// Start All Core M4 Programing /////////////////////

#ifdef CORE_CM4 

int  myCount=48;  //48 = ascii 0,    58 ascii 9

UART myUART3(PA_9,  PA_10, NC, NC); //TX, TR, RTS, CTS  NOTE: NC means not connected

void setup() { 
    pinMode(LEDB, OUTPUT);   // LEDB = blue, LEDG or LED_BUILTIN = green, LEDR = red 
   // Serial.begin(115200);
    myUART3.begin(9600);   

}

void loop() {
    myCount++;
   if (myCount >= 58){myCount = 48;}   //48 = ascii 0,    58 ascii 9
   digitalWrite(LEDB, !digitalRead(LEDB));   //switch on / off
   char x = (char)myCount;  //random(48, 57);  // ascii code for numbers 0 to 9
   myUART3.write(x); 
   delayMicroseconds(2);  // start at 100000 and check the order. Mine still fine at 2 microseconds
  
    
}

#endif

//////////////////// End All Core M4 Programing /////////////////////

```

hpssjellis, 2021-06-22:
I think I will close this issue. My best RPC using a struct program is here, but @facchinm it is very dependent on the delay(50); before the RPC.call, not sure why before is more important than after. Also it does not die gracefully it either works or completely fails.

If left running it eventually crashes but I think that is more to do with sending too much data to the Serial port.

At least RPC can be used with a struct and works, even if it is slow. (50 milliseconds) 

```
#ifdef CORE_CM7    // Start M7 Core programming

#include "RPC_internal.h"  // comes with the Arduino mbed core installation

// Define the structure
struct myStruct {
  bool  myNewEntry;
  int   x;
  float y;
  char  z;
  char  myCharArray[6];
};

//Activate the structure
myStruct myTestStruct;  // = {true, 5, 3.4, 'H','e','l','l','o'};

// Procedure to change the structure
int setVar(int a) {
    myTestStruct.x = (int)a;
    myTestStruct.myNewEntry   = true;
    return a;
  }

void setup() {
   RPC1.begin();
   bootM4();  
   Serial.begin(115200);

   // Allow the M4 core to change the structure
   RPC1.bind("setVar", setVar); // do these have to be the same?      
}

void loop() {
    if (myTestStruct.myNewEntry){
       Serial.println(String(myTestStruct.x));
       myTestStruct.myNewEntry = false;
    }
    
}

#endif  // Done M7 core programming

// ------------------------------------------------------------------------

#ifdef CORE_CM4  // Start M4 core programming

#include "RPC_internal.h"  // comes with the mbed board installation

int myCount=0;

void setup() {
   RPC1.begin();         
}

void loop() {
   myCount += 1;
   if (myCount >= 10000) {
      myCount = 0;
   }
   
                // important for this delay to be before the RPC call ??
                // Does not gracefully die, either works or doesn't
   delay(50);   // delay(50); CAREFUL, THIS DELAY MUST BE ENOUGH for the RPC
   auto res = RPC1.call("setVar", myCount).as<int>();
             
}

#endif  // end M4 Core Programming

```

closed 2021-06-22 by hpssjellis