arduino/ArduinoCore-mbed #239: Feature Request: LittleFS support for RP2040-based boards

khoih-prog, 2021-05-27:
IMHO, everybody loves to have **LittleFS** support for Nano_RP2040_Connect as well as other Arduino RP2040-based boards.

LittleFS has been supported in other [**RP2040 arduino-pico core**](https://github.com/earlephilhower/arduino-pico) as [**LittleFS** Library](https://github.com/earlephilhower/arduino-pico/tree/master/libraries/LittleFS) and is working OK.

The LittleFS feature is necessary for many WiFiManager-like libraries.

Best Regards,

facchinm, 2021-05-27:
Hi @khoih-prog ,
LittleFS is already supported, there are just no examples showing it :slightly_smiling_face: 
On a Pico, for example, you can use the second half of the flash as a LittleFS partition using some code like this
```
#include "FlashIAPBlockDevice.h"
#include "LittleFileSystem.h"
#include "mbed.h"

// 1MB partition, starting 1MB into the flash
FlashIAPBlockDevice bd(XIP_BASE + 0x100000, 0x100000);
mbed::LittleFileSystem fs("ota");

void setup() {
   int err = fs.mount(&bd);
   if (err) {
      fs.reformat(&bd);
   }
   // access the filesystem using normal POSIX APIs (fopen, fread) or mbed FileSystem APIs (https://os.mbed.com/docs/mbed-os/v6.10/apis/file-system-apis.html)
}

void loop() {}
```

closed 2021-05-27 by facchinm

khoih-prog, 2021-05-29:
Hi @facchinm 

I already tested the **LittleFS** and everything is OK by using POSIX APIs, just different from other LittleFS.

I've now used LittleFS successfully is my 3 libraries [DoubleResetDetector_Generic](https://github.com/khoih-prog/DoubleResetDetector_Generic), [MultiResetDetector_Generic](https://github.com/khoih-prog/MultiResetDetector_Generic) and [Ethernet_Manager](https://github.com/khoih-prog/Ethernet_Manager). Many more to be updated.

Thanks for your great works,