arduino/ArduinoCore-mbed #241: nano33ble

mikeysklar, 2021-05-27:
Multiple users of the Nano 33 BLE board have been having problems building any code that uses the NeoPixel library. We have narrowed down the issue to a change made to the 1.3.0 --> 1.3.1 release of the ArduinoCore-mbed board library. It looks like the -beta definitions were dropped for this board in that update and didn't get defined quite right in the 1.3.1 and subsequent releases.

The error that occurs on 1.3.1 and later is:

`/Users/mikeysklar/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':
/Users/mikeysklar/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:1528:45: error: invalid use of incomplete type 'PinDescription {aka struct _PinDescription}'
     pwm->PSEL.OUT[0] = g_APinDescription[pin].name;
                                             ^
In file included from /Users/mikeysklar/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.h:41:0,
                 from /Users/mikeysklar/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:46:
/Users/mikeysklar/Library/Arduino15/packages/arduino/hardware/mbed/2.1.0/cores/arduino/Arduino.h:83:16: note: forward declaration of 'PinDescription {aka struct _PinDescription}'
 typedef struct _PinDescription PinDescription;
                ^~~~~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Nano 33 BLE.`

mikeysklar, 2021-05-27:
Adafruit forum discussion: 

https://forums.adafruit.com/viewtopic.php?f=47&t=177925

jiaxiaochong, 2021-06-05:
cool

closed 2021-06-28 by facchinm