arduino/ArduinoCore-mbed #243: Raspberry Pi Pico board multicore example - not finding queue library functions (getting a undefined reference to)

Gerriko, 2021-05-28:
I'm using a Raspberry Pi Pico board. I am trying to get the **multicore_runner_queue.c** example (from the official Raspberry Pi Pico C SDK examples set) working using the Arduino IDE.

My headers are as follows:

```
#include <stdio.h>
#include "pico/util/queue.h"
#include "pico/multicore.h"

```

I initially had problems with this typedef

```
typedef struct
{
    void *func;
    int32_t data;
} queue_entry_t;

```

The Arduino compiler found a mismatch with `void *func` and what was used in the code. It also did not like` int32_t (*func)() = (int32_t(*)())(entry.func);`

So I amended code so that

```
typedef struct
{
    int32_t (*func)(int32_t);
    int32_t data;
} queue_entry_t;

```

and used `int32_t (*func)(int32_t) = (int32_t(*)(int32_t))(entry.func);
`

**Now I am getting an error where it is telling me:**

```
undefined reference to `queue_remove_blocking(queue_t*, void*)'
undefined reference to `queue_add_blocking(queue_t*, void*)'

```

These functions are used in both core0 and core1. For reference, the code snippet for core1 is:

```
void core1_entry() {
    
  while (1) {
    // Function pointer is passed to us via the queue_entry_t which also
    // contains the function parameter.
    // We provide an int32_t return value by simply pushing it back on the
    // return queue which also indicates the result is ready.

    queue_entry_t entry;

    queue_remove_blocking(&call_queue, &entry);

    int32_t (*func)(int32_t) = (int32_t(*)(int32_t))(entry.func);
    int32_t result = (*func)(entry.data);

    queue_add_blocking(&results_queue, &result);    
  } 
}

```

loulansuiye, 2021-07-05:
some header files have not inlcude in the Scons Scripts ,so it report errors as aboving

loulansuiye, 2021-07-05:
RT-thread has the same errors too