arduino/ArduinoCore-mbed #244: Nano 33 BLE digitalPinToPinName undefined reference

Klaus-KK, 2021-05-29:
I wrote a few examples to answer questions in the forum for the Arduino Nano 33 BLE. These examples no longer compile with the new version supplied with the Mbed Os Nano Boards (2.1.0) but work with version 1.3.2. The error points to the digitalPinToPinName function.

Here is a short example that creates PWM signal at pin 10.
```
#include "mbed.h"

#define PWM_PIN               10
#define PWM_FREQUENCY         32000

mbed::PwmOut pwmPin( digitalPinToPinName( PWM_PIN ) );

void setup()
{
  pwmPin.period( 1.0 / PWM_FREQUENCY );
  pwmPin.write( 0.5 );
}

void loop()
{
}
```

Here is the error message displayed
```
.. .ino.cpp.o: In function `__static_initialization_and_destruction_0':
... .ino:6: undefined reference to `digitalPinToPinName(unsigned char)'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Nano 33 BLE.
```

I looked at the source code and found the function was previously implemented in macros. h as macro. Now they are moved to pinDefinitions.h and the digitalPinToPinNames has been re-implemented as inline function. It did not look like a simple copy/past or typo error. There seems to be something additionally broken.

[https://github.com/arduino/ArduinoCore-mbed/blob/master/cores/arduino/pinDefinitions.h#L22](url)

Here is the link to the forum post where this was first discovered.
[https://forum.arduino.cc/t/arduino33-ble-external-interrupts/864391/9](url)

cattledogGH, 2021-05-30:
There is something broken with the links provided by Klaus-KK
Here they are again
[https://github.com/arduino/ArduinoCore-mbed/blob/master/cores/arduino/pinDefinitions.h](url)
[https://forum.arduino.cc/t/arduino33-ble-external-interrupts/864391/9](url)

cattledogGH, 2021-05-30:
There is something broken with the links provided by Klaus-KK. Here they are again

[https://github.com/arduino/ArduinoCore-mbed/blob/master/cores/arduino/pinDefinitions.h](url)

[https://forum.arduino.cc/t/arduino33-ble-external-interrupts/864391/9](url)

andrew-sz, 2021-06-06:
Hey all, it seems that `digitalPinToPinName` function is completely broken.   However, after reviewing the mbed source was able to assign pins by using the canonical pin name (i.e. P0_13).   

Possible workaround until bsp is fixed is to use the pin names in PinNames.h as this code will compile.  

```
#include "mbed.h"
#define PWM_FREQUENCY         32000

mbed::PwmOut pwmPin(P0_10);

void setup()
{
  pwmPin.period( 1.0 / PWM_FREQUENCY );
  pwmPin.write( 0.5 );
}

void loop()
{
}

facchinm, 2021-06-07:
Hi,
the `digitalPinToPinName` is not broken, is just hidden (private) by design. This is necessary because otherwise some very heavy headers from mbed would be included by any file including `Arduino.h` making the compilation painfully slow, especially on Windows.
If you need that function, the correct way is to include `pinDefinitions.h` in the sketch or, better, the library.

closed 2021-06-07 by facchinm