arduino/ArduinoCore-mbed #250: Missing time.c and stdio.h files

khoih-prog, 2021-06-02:
I'm trying this simple timer-related code, which is simplified from [RPI pico SDK **hardware_timer**](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__timer.html)

```
//#include "pico/stdlib.h"
#include "pico/time.h"

volatile bool timer_fired = false;

int64_t alarm_callback(alarm_id_t id, void *user_data) 
{
  Serial.print("Fired. Timer");
  Serial.println((int) id);
  
  timer_fired = true;
  // Can return a value here in us to fire in the future
  return 0;
}

void setup()
{
  Serial.begin(115200);
  while (!Serial);

  Serial.println("TimerTest");

  // Call alarm_callback in 2 seconds
  add_alarm_in_ms(2000, alarm_callback, NULL, false);

  // Wait for alarm callback to set timer_fired
  while (!timer_fired) 
  {
    tight_loop_contents();
  }

  Serial.println("Done");
}

void loop() 
{
}
```

This code compiles OK using SDK or [**RP2040 arduino-pico core**](https://github.com/earlephilhower/arduino-pico)

But there are some issues with this core

1. Missing `stdio.h`

In [stdlib.h#L10-L12](https://github.com/arduino/ArduinoCore-mbed/blob/master/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_stdlib/include/pico/stdlib.h#L10-L12)

```
#include "pico.h"
#include "pico/stdio.h"
#include "pico/time.h"
```

but there is no `stdio.h` in the core

2. Missing `time.c`

Comment out

```
#include "pico/stdlib.h"
```

The compiler also generates error with this core, possibly no `time.c` was included in the core

```
/home/kh/Arduino/Testing/RPi_Pico/Testing/Mbed_TimerInterrupt/Stdio_Issue/Stdio_Issue.ino: In function 'void setup()':
Stdio_Issue:24:3: error: 'add_alarm_in_ms' was not declared in this scope
   add_alarm_in_ms(2000, alarm_callback, NULL, false);
   ^~~~~~~~~~~~~~~
/home/kh/Arduino/Testing/RPi_Pico/Testing/Mbed_TimerInterrupt/Stdio_Issue/Stdio_Issue.ino:24:3: note: suggested alternative: 'alarm_id_t'
   add_alarm_in_ms(2000, alarm_callback, NULL, false);
   ^~~~~~~~~~~~~~~
   alarm_id_t
exit status 1
'add_alarm_in_ms' was not declared in this scope
```

In [time.h#L548-L549](https://github.com/arduino/ArduinoCore-mbed/blob/master/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_time/include/pico/time.h#L548-L549), the function `add_alarm_in_ms` is declared as

```
static inline alarm_id_t add_alarm_in_ms(uint32_t ms, alarm_callback_t callback, void *user_data, bool fire_if_past) {   
  return alarm_pool_add_alarm_in_ms(alarm_pool_get_default(), ms, callback, user_data, fire_if_past);}
```

Regards,

khoih-prog, 2021-06-02:
I tried to change in [defines.txt#L28](https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/RASPBERRY_PI_PICO/defines.txt#L28) to

```
-DPICO_TIME_DEFAULT_ALARM_POOL_DISABLED=0
```

but still not OK yet.

Error

```
sketch/Stdio_Issue.ino.cpp.o: In function `add_alarm_in_ms':
/home/kh/.arduino15/packages/arduino/hardware/mbed_rp2040/2.1.0/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_time/include/pico/time.h:549: undefined reference to `alarm_pool_get_default'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Raspberry Pi Pico.

```

ernesto75, 2021-06-17:
Confirmed, i cannot change the cpu clock speed on the pico with this simple sketch because of that error
`#include "pico/stdlib.h"
void setclock_khz(uint32_t khz) {
  
set_sys_clock_khz(khz,true);
}

int main() {
sleep_ms(3000);
printf("setting cpu clock to: %d Khz \n", 50000);
set_sys_clock_khz(50000,true);
}`

Arduino: 1.8.15 (Windows 10), Board: "Raspberry Pi Pico"

In file included from C:\Users\Ernesto\AppData\Local\Temp\arduino_modified_sketch_761225\sketch_jun17b.ino:1:0:

C:\Users\Ernesto\AppData\Local\Arduino15\packages\arduino\hardware\mbed_rp2040\2.1.0\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_stdlib/include/pico/stdlib.h:11:10: fatal error: pico/stdio.h: No such file or directory

 #include "pico/stdio.h"

          ^~~~~~~~~~~~~~

compilation terminated.

exit status 1

Error compiling for board Raspberry Pi Pico.

Dresch123, 2021-09-06:
I concur. Without the stdio.h file there does not seem to be any way to set the system clock with the Pico Pi. I posted this question a week ago but see that khioh-prog and ernesto75 posted it back in June. This seems like a handicap to the Mbed implementation.

Is there anyway to get the attention of  the Mbed core developers concerning this? My attempts at adding the stdio.h from the Earle core have not been successful...
Thanks,
Bill