arduino/ArduinoCore-mbed #252: Missing USB Keyboard and Mouse support

blackketter, 2021-06-02:
The USB keyboard and mouse examples do not build for RP2040 based boards with the error:

```
In file included from /Applications/Arduino.app/Contents/Java/examples/09.USB/Keyboard/KeyboardMessage/KeyboardMessage.ino:23:0:
/Applications/Arduino.app/Contents/Java/libraries/Keyboard/src/Keyboard.h:25:10: fatal error: HID.h: No such file or directory
 #include "HID.h"
          ^~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Nano RP2040 Connect.
```

Is this feature expected to work?

facchinm, 2021-06-07:
Hi @blackketter ,
there's still a bit of incompatibility between this core and any PluggableUSB derived library, so plain old Keyboard and Mouse won't work.
In the meantime, you can use `PluggableUSBHID` from here https://github.com/arduino/ArduinoCore-mbed/tree/master/libraries/USBHID (it's included in the core) with something like this sketch

```c++
#include "PluggableUSBHID.h"
#include "USBKeyboard.h"

USBKeyboard Keyboard;

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
  Keyboard.printf("Hello world\n\r");
}
```

beyarkay, 2022-03-13:
*Warning* for future users looking at this, make sure the delay is over 10 seconds or so. If it's at the default 1 second, then you don't have enough time between the newlines to go through and upload new code to the Nano, effectively bricking the device.

dogtopus, 2022-04-07:
> there's still a bit of incompatibility between this core and any PluggableUSB derived library, so plain old Keyboard and Mouse won't work.

Will it move to the standard PluggableUSB API in the future? I'm currently working on a USB related library and it would make supporting standard PluggableUSB infinitely more attractive since I'm mostly targeting Arduino-mbed right now but it would be nice to also include support down to AVR Arduino with minimal effort.