arduino/ArduinoCore-mbed #254: Digital Pin defines missing in RP2040 Connect

WayneKeenan, 2021-06-05:
Why doesn't the RP2040 `pins_arduino.h` contain the digital pin defines like the 33BLE?

e.g. shouldn't this be in `.../mbed_nano/2.1.0/variants/NANO_RP2040_CONNECT/pins_arduino.h` ?

```
// Digital pins
// -----------
#define D0  (0u)
#define D1  (1u)
#define D2  (2u)
#define D3  (3u)
#define D4  (4u)
#define D5  (5u)
#define D6  (6u)
#define D7  (7u)
#define D8  (8u)
#define D9  (9u)
#define D10 (10u)
#define D11 (11u)
#define D12 (12u)
#define D13 (13u)
```

WanaGo, 2021-09-08:
Wanting to know this too
How do you know what the GPIO pins are called in the IDE?
I cant find the document which explains what term you use in the IDE to associate to each of the GPIO on the Pi Pico... 
Please help

vbargsten, 2024-01-09:
Was also missing the definitions for Dx pins.

+1