arduino/ArduinoCore-mbed #255: Nano33ble - undefined reference to "SDBLockDevice.h"

Cdo-Github, 2021-06-06:
I would like to enable ble bonding. An SDcard is attached and a mbed_app.json file is included. When I try to compile in Platformio (updated) there is always an "undefined reference to SDBLockDevice.h" error. 

Does someone have a solution for my problem?

facchinm, 2021-06-07:
Hi @Cdo-Github ,
the SDBlockDevice code is not being compiled by default for Nano33BLE target, so to use it you can either:
* add the mbed_app modification here https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/ARDUINO_NANO33BLE/conf/mbed_app.json , then run `./mbed-os-to-arduino -a ARDUINO_NANO33BLE:ARDUINO_NANO33BLE` to rebuild the mbed library (only works from Linux ATM)
* or copy the source from https://github.com/arduino/mbed-os/blob/master/storage/blockdevice/COMPONENT_SD/source/SDBlockDevice.cpp into your sketch, changing the include to `#include "COMPONENT_SD/include/SD/SDBlockDevice.h"`

Cdo-Github, 2021-06-25:
Sorry for my late reply. Now, it compiled but my SDcard reader seems to be broken.

Thank you very much!

closed 2021-06-28 by facchinm