arduino/ArduinoCore-mbed #267: Feature Request: SoftwareSerial for the Portenta

hpssjellis, 2021-06-19:
## Feature Request: SoftwareSerial for the Portenta

I have been trying to find the library SoftwareSerial so that I can use more than one UART channel with the base Portenta board. I found softwareSerial as a core [here](https://github.com/arduino/ArduinoCore-avr/tree/master/libraries/SoftwareSerial) for the AVR boards. Is there a workaround for MBED boards, or can SoftwareSerial be prted to the Portenta. I tried just copying the library to the MBED core but it fails on 

```ResolveLibrary(util/delay_basic.h```

which I can't find. I will look into an MBED solution.

hpssjellis, 2021-06-19:
Found and testing this library

https://github.com/PaulStoffregen/SoftwareSerial

sidprice, 2021-06-19:
Can you not use the other hardware UARTS on the Portenta H7?

hpssjellis, 2021-06-19:
@sidprice   As far as I know the other UART's are only available using the very large breakout board. Do you have any suggestions.

sidprice, 2021-06-19:
I use the Portenta Breakout board. https://store.arduino.cc/usa/portenta-breakout

hpssjellis, 2021-06-20:
@sidprice Serial communication between cores works great with the Portenta Breakout board, but I want it to work without the breakout board, the main issue is posted [here issue 266](https://github.com/arduino/ArduinoCore-mbed/issues/266), where for simple communication Serial1/UART1 can be used to communicate between cores, but for complex communication it breaks. For complex communication you can use  2 different UARTS to communicate between cores but that forces you to have the breakout board. I would like a solution that can be used for regular Portenta users and SoftwareSerial would be an easy solution. I will look into an MBED version of SoftwareSerial. Does anyone have a link to follow.

hpssjellis, 2021-06-20:
@facchinm with the main board only having one UART1 on D14, D13 would it make sense to be able to run a version of SoftwareSerial on the MBED boards?

This topic is related to [https://github.com/arduino/ArduinoCore-mbed/issues/266](https://github.com/arduino/ArduinoCore-mbed/issues/266)

hpssjellis, 2021-06-22:
Closing this as the breakout board solves the problem

closed 2021-06-22 by hpssjellis