arduino/ArduinoCore-mbed #268: mbed Timeout not working on Nano 33 ble board

JAICHANGPARK, 2021-06-22:
Hi all

I tried to use mbed timeout api in my test code
but time out not works for me 

```
#include "mbed.h"

mbed::Timeout flipper;

void flip() {
   Serial.println("flip");
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  while(!Serial){}
  Serial.println("hello world");
  
  flipper.attach(&flip, 5.0); // setup flipper to call flip after 2 seconds
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

## result 
- output only hello world
[image]

- how to solve this problem?

manchoz, 2021-06-22:
Hi @JAICHANGPARK,
In the callback for timers, timeouts and the like, you can call only re-entrant functions and functions that do not do I/O. Unfortunately, Serial.print() is neither re-entrant nor "pure" (in a Functional way). Those callbacks need to be managed as ISRs, using `volatile`s and simple code.

A better approach would use the callback to set a flag and then check for that flag in the loop for executing your non-re-entrant functions.

A more complex approach would use mbedOS' [EventQueue](https://os.mbed.com/docs/mbed-os/v6.11/apis/eventqueue.html)s.

closed 2021-07-09 by facchinm

Gerriko, 2022-04-03:
As @manchoz explains. Also need to use rtos.h and a thread.

```
#include "mbed.h"
#include "rtos.h"
#include <EventQueue.h>

events::EventQueue queue(32 * EVENTS_EVENT_SIZE);

mbed::Timeout flipper;

rtos::Thread t;

void print_handler() {
  Serial.println("flip");
}

void flip() {
  queue.call(print_handler);
}

void setup() {
  // put your setup code here, to run once:
  t.start(callback(&queue, &events::EventQueue::dispatch_forever));
  Serial.begin(9600);
  while(!Serial){}
  Serial.println("hello world");
  
  flipper.attach(&flip, 2.0); // setup flipper to call flip after 2 seconds
}

void loop() {
  // put your main code here, to run repeatedly:

}
```