arduino/ArduinoCore-mbed #272: Issue with BufferedSerial on RP2040 platfrom

ahmedalkabir, 2021-07-02:
hey.
I faced an issue when using BufferedSerial class specially when writing buffer or data, it just write 2 data and stop ??, actually it is blocking and stuck.

here's the code 
```c++
#include <Arduino.h>
#include "mbed.h"

rtos::Thread thread_1;
static mbed::BufferedSerial serial_port(p4, p5);

void uart_thread()
{
  serial_port.set_baud(115200);
  serial_port.write("buffer", 7); // never write all of 
  serial_port.write("9", 1);
  char buf[32] = {0};
  while (1)
  {
    if (uint32_t num = serial_port.read(buf, sizeof(buf)))
    {
      serial_port.write(buf, num);
    }
  }
}

void setup()
{
  thread_1.start(uart_thread);
}
```

I read Serial source code and I noticed it depends on UnbufferedSerial, I tested UnbufferedSerial and it worked as I expected but BufferedSerial's write functionally doesn't work well !

I used PlatformIO and Raspberry Pi Pico which is based on RP2040, so I hope the issue resolve as soon as possible and thanks  😁

magnusnordlander, 2021-07-03:
I'm having a similar issue with an Arduino Nano RP2040 Connect; I'm assuming my issue is related.

facchinm, 2021-07-05:
Hi there,
thanks for reporting the issue!
The problem seems to be originated from https://github.com/arduino/mbed-os/blob/d7ccde9f58b366d8653e189ba698c996d4cfa3da/targets/TARGET_RASPBERRYPI/TARGET_RP2040/serial_api.c#L105-L113 . We only provide RX interrupts capability at the moment, so the TX interrupt used by BufferedSerial is not operational.

From the datasheet, it looks like `UARTRXINTR` and `UARTTXINTR` can be used to discriminate which event happened, but fixing this would require a not-so-small refactoring. 
If anyone's interested in taking a look, it would be a wonderful contribution :slightly_smiling_face: