arduino/ArduinoCore-mbed #273: nano 33 ble for nRF52833

UGMWJ, 2021-07-02:
Hi,

I have build my custom design based on nano 33 ble but couldn't fabricate the design as  NINA-B306-00B is out of stock. I thought to use NINA-B400-00B if I can  compile Arduino library for nRF52833 mcu to replace current nRF52840 library. 

Let me know how I can compile nano 33 BLE for new MCU.

Many thanks and kind regards,
Ghalib

facchinm, 2021-07-05:
Hi @UGMWJ ,
from [this table](https://www.nordicsemi.com/-/media/Publications/WQ-Product-guide/Wireless-Q---Q1---Product-Guide.pdf?la=en&hash=55195302B5331ECA0F7D841B58D7F97DCC945F26) it looks like nRF52833 and nRF52840 have the same peripheral set except QSPI and Cryptocell.  nRF52833 is not yet supported on mbed-os but I think the nRF52840 (nano33ble) build should work to a certain extent.

When it will be supported, you'll be able to compile `libmbed` for your target by running
```./mbed-os-to-arduino $TARGET_NAME:$VARIANT_NAME``` and then creating the relevant entries in `boards.txt`.

Let me know if it works when setting Nano 33 BLE as target; I'm closing the issue in the meantime.

closed 2021-07-05 by facchinm

alvee1994, 2021-08-29:
hi @UGMWJ ,

Wow I am in the same position now. did you get your device to work?

ghalibjanjua, 2021-08-30:
Hi @alvee1994 

I did tired it as I was able to get NINA-B306-00 boards. Best of luck for your work.

ghalibjanjua, 2021-08-30:
Many thanks @facchinm 

G