arduino/ArduinoCore-mbed #279: Does Arduino Nano RP2040 library for Arduino IDE support SPI?

AtaKhiabani, 2021-07-06:
I am trying to use the Arduino Nano RP2040 as a slave and I have installed the "Arduino mbed" library, but it looks like in Arduino IDE the SPI registers such as "SSPCR0" are not recognized. Is there any example for this board using SPI as slave?

facchinm, 2021-07-09:
Hi @AtaKhiabani ,
the Nano RP2040 supports SPI (in master mode only at the moment). The development is still ongoing in https://github.com/ARMmbed/mbed-os/pull/14488 but you can probably just use pico-sdk functions (which are exported but untested).

AtaKhiabani, 2021-07-09:
> Hi @AtaKhiabani ,
> the Nano RP2040 supports SPI (in master mode only at the moment). The development is still ongoing in [ARMmbed/mbed-os#14488](https://github.com/ARMmbed/mbed-os/pull/14488) but you can probably just use pico-sdk functions (which are exported but untested).

Thank you for your response. I actually need to use Nano RP2040 in slave mode. I will try using the pico-sdk functions though.