arduino/ArduinoCore-mbed #280: 2.2 breaks compatibility with WifiNINA library

frutabruta, 2021-07-08:
Hello,
since core version 2.2.0 sketches using WifiNINA library (version 1.8.10) cannot be compiled for Arduino Nano RP2040 nor Nano 33 BLE in Arduino IDE 1.8.15 (on Windows 10). The same sketch (ScanNetworksAdvanced) can be compiled using ArduinoCore-mbed version 2.1.0 without any problems.
The error is following:
`
C:\Users\adam\Documents\Arduino\libraries\WiFiNINA\src\utility\spi_drv.cpp: In static member function 'static void SpiDrv::begin()':
C:\Users\adam\Documents\Arduino\libraries\WiFiNINA\src\utility\spi_drv.cpp:108:15: error: 'NINA_GPIO0' was not declared in this scope
       pinMode(NINA_GPIO0, OUTPUT);
               ^~~~~~~~~~
C:\Users\adam\Documents\Arduino\libraries\WiFiNINA\src\utility\spi_drv.cpp:108:15: note: suggested alternative: 'NINA_GPIOIRQ'
       pinMode(NINA_GPIO0, OUTPUT);
               ^~~~~~~~~~
               NINA_GPIOIRQ
C:\Users\adam\Documents\Arduino\libraries\WiFiNINA\src\utility\spi_drv.cpp: In static member function 'static int SpiDrv::available()':
C:\Users\adam\Documents\Arduino\libraries\WiFiNINA\src\utility\spi_drv.cpp:65:25: error: 'NINA_GPIO0' was not declared in this scope
 #define NINA_GPIOIRQ    NINA_GPIO0
                         ^
C:\Users\adam\Documents\Arduino\libraries\WiFiNINA\src\utility\spi_drv.cpp:602:25: note: in expansion of macro 'NINA_GPIOIRQ'
     return (digitalRead(NINA_GPIOIRQ) != LOW);
                         ^~~~~~~~~~~~
C:\Users\adam\Documents\Arduino\libraries\WiFiNINA\src\utility\spi_drv.cpp:65:25: note: suggested alternative: 'NINA_GPIOIRQ'
 #define NINA_GPIOIRQ    NINA_GPIO0
                         ^
C:\Users\adam\Documents\Arduino\libraries\WiFiNINA\src\utility\spi_drv.cpp:602:25: note: in expansion of macro 'NINA_GPIOIRQ'
     return (digitalRead(NINA_GPIOIRQ) != LOW);
                         ^~~~~~~~~~~~
exit status 1
`

Should this be published to arduino-libraries/WiFiNINA too?

per1234, 2021-07-09:
Hi @frutabruta. You are using an outdated version of the WiFiNINA library. Please update to the latest version 1.8.12 to fix this error. You can do that via the Arduino IDE's Library Manager (**Sketch > Include Library > Manage Libraries...**).

frutabruta, 2021-07-09:
Hi @per1234 , you are right, the new WiFiNINA is compatible with Core 2.2.0, my bad.

closed 2021-07-09 by frutabruta