arduino/ArduinoCore-mbed #289: PDM Library - Random Microphone Spikes on Nano RP2040 Connect

julienmartin84, 2021-07-18:
Hello!
I have been playing with the PDM library and the bundled PDM example on an Arduino Nano RP2040 Connect.

**v2.1.0**: no spike artifact
**v2.2.0**: spikes artifacts
**v2.3.1**: spikes artifacts

Screenshot 1: no noise in the room
[image]

Screenshot 2: was playing a clean sine wave near the arduino
[image]

The spikes seem to disappear when using a frequency of 8kHz or less instead of the 16kHz default of the example.
I describe the problem in details [here](https://forum.arduino.cc/t/nano-rp2040-connect-microphone-spike/881069/12)
In this Arduino forum topic, user Klaus_K identified some change that occurred with 2.2.0 and that could maybe cause the issue.

facchinm, 2021-08-10:
Hi @julienmartin84 ,
thanks for taking the time to report the issue here.
The glitch is due to the new clock frequency provided to the microphone (starting from https://github.com/arduino/ArduinoCore-mbed/commit/c97ec7e540be9a602b225309229d2a1e3366e937), which is above the nominal maximum at 16khz sampling rate
[image]
I pushed #302 to fix this, it would be awesome if you could test it and report if it really fixes the issue :wink:

closed 2021-08-11 by facchinm

julienmartin84, 2021-08-14:
It is working correctly after i updated **Arduino Mbed OS Nano Boards**
Thank you!

RickorDD, 2022-08-15:
I have a problem with my Arduino Connect PR2040 and the internal MIC.
SerialPlotterExample and Edge Impulse runs not with Decimation = 64 only with 128 but there the Spikes.

https://forum.edgeimpulse.com/t/arduino-nano-rp2040-help/4638/22