arduino/ArduinoCore-mbed #290: How to access both sd cards on Portenta Breakout and Vision shields?

hpssjellis, 2021-07-19:
I have been writing some sd card code and it occured to me that the Portenta Breakout board can have a vision shield attached making access to 2 sd cards. Can they both be accessed or are they considered the same card reader?

JSON and CSV examples [here](https://github.com/hpssjellis/portenta-pro-community-solutions/tree/main/examples/e-portenta-machine-learning/e-a-portenta-edge-impulse-ml/e-a-a-vision-shield)

Here is some simple Portenta sd card .txt file writing code: How would I attempt to direct it to the breakout board, or the vision shield?
```
/*
  Portenta - TestSDCARD

  The sketch shows how to mount an SDCARD and list its content.
  then add a file.

  Note: SD Card seems finisky, takes a few times to load the filesystem

  The circuit:
   - Portenta H7 + Vision Shield
   or
   - Portenta H7 + Portenta Breakout

  This example code is in the public domain.
*/
#include "SDMMCBlockDevice.h"
#include "FATFileSystem.h"

SDMMCBlockDevice block_device;
mbed::FATFileSystem fs("fs");

void setup() {
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
  
  delay(5000); // time to connect serial
  Serial.println("Connect serial if needed");
  delay(5000);
  //while (!Serial);  // blocking call

  Serial.println("Mounting SDCARD...");
  int err =  fs.mount(&block_device);
  if (err) {
     Serial.println("No SD Card filesystem found, please check SD Card on computer and manually format if needed.");
     // int err = fs.reformat(&block_device);  // seriously don't want to format your good data
  }

}

void loop() {
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN)); // flips LED on and off
  
  // Make a folder if needed
  mkdir("fs/myFolder2",0777);                     // 0777 full access permissions linux style 
   
  char myFileName[] = "fs/myFolder2/test2.txt";   // "fs/" needs to be there, think fileSystem
 
  FILE *myFile = fopen(myFileName, "w");          // "a" for append (add to file), "w" write, "r" read ??
 
     Serial.println(myFileName);
     fprintf(myFile,"Test how cool this is \r\n");
     fprintf(myFile,"Also this line \r\n");
   
  fclose(myFile);

  Serial.println("------------------------- Done Writing file --------------------------------");
  delay(10);
   
  unsigned char c; 
  FILE *fp = fopen(myFileName, "r");              // "r" read only
     while (!feof(fp)){                           // while not end of file
        c=fgetc(fp);                              // get a character/byte from the file
        //printf("Read from file %02x\n\r",c);    // and show it in hex format
        Serial.print((char)c);                    // show it as a text character
     }
  fclose(fp); 
  Serial.println("------------------------- Done Showing file --------------------------------");

  delay(10000);   // wait a bit
  
}

```

pennam, 2021-07-20:
Hi @hpssjellis personally i've never considered the option to connect a vision shield to a portenta breakout, but i would say they are considered the same SDCARD since they are sharing the same PORTENTA pins and so it is not possible to direct the access to one specific SDCARD holder. They can both be used, but only connecting ONE SDCARD at time.

hpssjellis, 2021-07-20:
Your right @pennam with vision shield attached to the Portenta Breakout board both sd card locations work fine individually. 
What is interesting is (I tried a 32Gb and 64 GB sd card) if you have 2 sd cards inserted, for me it picked the vision shield over the breakout board. Everything worked fine.

Gotcha's:  
1. The Portenta really likes to format cards that have been changed by the windows system (that is why I have the format line of code commented out)
2. My old vision shield is finicky, works better with a bit of paper inserted in the sd card holder to push the card onto the contacts better, doesn't seem to be an issue with the breakout board or my new vision shields.

Going to close this. Thanks for the input @pennam 

```
/*
  Portenta - TestSDCARD

  The sketch shows how to mount an SDCARD and list its content.
  then add a file.

  Note: SD Card seems finisky, takes a few times to load the filesystem

  The circuit:
   - Portenta H7 + Vision Shield
   or
   - Portenta H7 + Portenta Breakout

  This example code is in the public domain.
*/
#include "SDMMCBlockDevice.h"
#include "FATFileSystem.h"

SDMMCBlockDevice block_device;
mbed::FATFileSystem fs("fs");
//FILE *myFile;
//char buffer[40];   // must be long enough for all data points with commas: 56,34,23,56

void setup() {
  Serial.begin(115200);
  while (!Serial);  // may want to delete this later

  Serial.println("Mounting SDCARD...");
  delay(10);
  int err =  fs.mount(&block_device);
  if (err) {
    // Reformat if we can't mount the filesystem
    // this should only happen on the first boot
    Serial.println("No filesystem found, please check on computer and manually format if needed.");
  //  err = fs.reformat(&block_device);  // seriously don't want to format your good data
  }
  if (err) {
     Serial.println("Error formatting SDCARD ");
     while(1);
  }
  
  DIR *dir;
  struct dirent *ent;
  int dirIndex = 0;

  Serial.println("List SDCARD content: ");
  if ((dir = opendir("/fs")) != NULL) {
    // Print all the files and directories within directory (not recursively)
    while ((ent = readdir (dir)) != NULL) {
      Serial.println(ent->d_name);
      dirIndex++;
    }
    closedir (dir);
  } else {
    // Could not open directory
    Serial.println("Error opening SDCARD\n");
    while(1);
  }
  if(dirIndex == 0) {
    Serial.println("Empty SDCARD");
  }

 
  
 Serial.println("------------------------- Done --------------------------------"); 
}

void loop() {
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN)); // flips LED on and off
  
  // Make a folder if needed
  mkdir("fs/myFolder2",0777);                     // 0777 full access permissions linux style 
   
  char myFileName[] = "fs/myFolder2/test2.txt";   // "fs/" needs to be there, think fileSystem
 
  FILE *myFile = fopen(myFileName, "w");          // "a" for append (add to file), "w" write, "r" read ??
 
     Serial.println(myFileName);
     fprintf(myFile,"Test how cool this is \r\n");
     fprintf(myFile,"Also this line \r\n");
   
  fclose(myFile);

  Serial.println("------------------------- Done Writing file --------------------------------");
  delay(10);
   
  unsigned char c; 
  FILE *fp = fopen(myFileName, "r");              // "r" read only
     while (!feof(fp)){                           // while not end of file
        c=fgetc(fp);                              // get a character/byte from the file
        //printf("Read from file %02x\n\r",c);    // and show it in hex format
        Serial.print((char)c);                    // show it as a text character
     }
  fclose(fp); 
  Serial.println("------------------------- Done Showing file --------------------------------");

  delay(10000);   // wait a bit
  
}

```

closed 2021-07-20 by hpssjellis