arduino/ArduinoCore-mbed #300: pico-sdk files missing extern "C" guard

josefwegner, 2021-08-03:
It looks like several files in **cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common** are missing an `extern "C"` block.

For example **rp2_common/hardware_rtc/include/hardware/rtc.h**.

This means that this simple program
```
#include "hardware/rtc.h"
void setup() {
  rtc_init();
}

void loop() {}
```
 fails to compile with following error:
```
$arduino-cli compile -b arduino:mbed_rp2040:pico
/tmp/arduino-sketch-54917AA5DAEB0A8E434881784695EAF6/sketch/clock_tm1637.ino.cpp.o: In function `setup':
/home/pi/Projects/arduino/clock_tm1637/clock_tm1637.ino:58: undefined reference to `rtc_init()'
collect2: error: ld returned 1 exit status

Error during build: exit status 1
```

Adding an
```
extern "C"
{
  void rtc_init           (void);
}
```
fixes the problem.

The file **hardware/rtc.h** included in ArduinoCore-mbed has no `extern "C"`. The same file in the pico-sdk has this definition.

There might be more files having this issue.

facchinm, 2021-08-04:
Hi @josefwegner , we used version 1.0.0 of pico-sdk for the initial bringup; it should be updated to a more recent one [here](https://github.com/ARMmbed/mbed-os/pull/14488) and the core recompiled accordingly. 
Will notify you as soon as this happens :slightly_smiling_face:

josefwegner, 2021-08-04:
Hi @facchinm, thank you for the explanation. Yes, the 1.0.0 version was missing the `extern "C"`. I just checked your draft, but it seems that the rtc.h there is still from 1.0.0.

facchinm, 2021-08-04:
Indeed, the PR should be updated before actually going live :wink:

fjansson, 2021-11-10:
Another issue I just met with Arduino for the Pico, mbed_rp2040/2.5.2/, is that some of the PWM functions have wrong types, signed instead of unsigned. This is confusing and doesn't match the current Pico SDK documentation. For example:
```
static inline int16_t pwm_get_counter(uint slice_num)   // old, in the Arduino core
static inline uint16_t pwm_get_counter(uint slice_num)  // newer, from pico-sdk version 1.1.0 
```

josefwegner, 2022-10-15:
Tested with latest ArduinoCore-mbed and RTC functions work out-of-the-box. All pico-sdk files now seem to have the correct `extern C` guards. Closing this issue.

closed 2022-10-15 by josefwegner