arduino/ArduinoCore-mbed #305: MBED FlashIAPBlockDevice API read/write problem with Block 1

Gerriko, 2021-08-16:
Hi @khoih-prog ,
LittleFS is already supported, there are just no examples showing it :slightly_smiling_face: 
On a Pico, for example, you can use the second half of the flash as a LittleFS partition using some code like this
```
#include "FlashIAPBlockDevice.h"
#include "LittleFileSystem.h"
#include "mbed.h"

// 1MB partition, starting 1MB into the flash
FlashIAPBlockDevice bd(XIP_BASE + 0x100000, 0x100000);
mbed::LittleFileSystem fs("ota");

void setup() {
   int err = fs.mount(&bd);
   if (err) {
      fs.reformat(&bd);
   }
   // access the filesystem using normal POSIX APIs (fopen, fread) or mbed FileSystem APIs (https://os.mbed.com/docs/mbed-os/v6.10/apis/file-system-apis.html)
}

void loop() {}
```

_Originally posted by @facchinm in https://github.com/arduino/ArduinoCore-mbed/issues/239#issuecomment-849403089_

I had spotted this suggestion, so I gave it a try and it works.
As I wanted to store bytes directly, I skipped the littlefs and focused on using mbed's FlashIAPBlockDevice API. The mbed website has an example (https://os.mbed.com/docs/mbed-os/v6.13/apis/flashiapblockdevice.html), which works but when you change block 0 to block 1 the first character printed is missing. So you get "ello World" instead of "hello World".

khoih-prog, 2021-08-16:
You can try my [LittleFS_Mbed_RP2040 library](https://github.com/khoih-prog/LittleFS_Mbed_RP2040)

---

## Features

This library is just a simple LittleFS wrapper to facilitate your usage of LittleFS for the onboard flash on **MBED RP2040-based boards such as Nano_RP2040_Connect, RASPBERRY_PI_PICO**, using [**Arduino-mbed RP2040** core](https://github.com/arduino/ArduinoCore-mbed)

The filesystem access uses normal [POSIX APIs](https://www.tutorialspoint.com/c_standard_library/stdio_h.htm) or [mbed FileSystem APIs](https://os.mbed.com/docs/mbed-os/v6.10/apis/file-system-apis.html)

Gerriko, 2021-08-16:
> You can try my [LittleFS_Mbed_RP2040 library](https://github.com/khoih-prog/LittleFS_Mbed_RP2040)
> ## Features
>
> This library is just a simple LittleFS wrapper to facilitate your usage of LittleFS for [...]

Thanks this looks very useful. I will try it out. I like the fact that you've tested with 64kBytes.

Gerriko, 2021-08-16:
Clearly the "thick block" sitting on my shoulders failed to understand the meaning of "block"... DOH.

closed 2021-08-16 by Gerriko

khoih-prog, 2021-08-30:
FYI, the new [**FS_Nano33BLE** Library](https://github.com/khoih-prog/FS_Nano33BLE) is a simple LittleFS/FATFS wrapper to facilitate your usage of LittleFS/FATFS for the onboard flash on **MBED nRF52840-based boards, such as Nano_33_BLE, Nano_33_BLE_Sense**, using [**Arduino-mbed mbed_nano** core](https://github.com/arduino/ArduinoCore-mbed).

It relies on the same MBED FlashIAPBlockDevice API.