arduino/ArduinoCore-mbed #309: Is it possible to overclock the Pico with ArduinoCore-mbed?

Dresch123, 2021-08-30:
Is it possible to change the system clock to overclock or underclock thru the ArduinoCore-mbed?
Thanks,
Bill

Dresch123, 2021-09-06:
#include <SPI.h>
#include "pico/stdlib.h"
#include "pico/stdio.h"
#include "hardware/vreg.h"
#include "hardware/clocks.h"
#include <stdlib.h>

void setup() {
  // put your setup code here, to run once:
  vreg_set_voltage(VREG_VOLTAGE_1_30);
  set_sys_clock_khz(140000,true); 
  Serial.begin(115200);
  pinMode(25, OUTPUT);
  digitalWrite(25, HIGH);
}

void loop() {
  // put your main code here, to run repeatedly:

}
This code compiles just fine on the Earle Philhower Core but not in the Mbed core. It appears that pico\stdio.h is not part of the Mbed package. Is there a fix for this?

tierneytim, 2021-09-06:
kind of, insert these functions at the top of your .ino file between your includes and setup() and it should work. 

```
#include "hardware/pll.h"
#include "hardware/clocks.h"

void set_sys_clock_pll(uint32_t vco_freq, uint post_div1, uint post_div2) {
  if (!running_on_fpga()) {
    clock_configure(clk_sys,
                    CLOCKS_CLK_SYS_CTRL_SRC_VALUE_CLKSRC_CLK_SYS_AUX,
                    CLOCKS_CLK_SYS_CTRL_AUXSRC_VALUE_CLKSRC_PLL_USB,
                    48 * MHZ,
                    48 * MHZ);

    pll_init(pll_sys, 1, vco_freq, post_div1, post_div2);
    uint32_t freq = vco_freq / (post_div1 * post_div2);

    // Configure clocks
    // CLK_REF = XOSC (12MHz) / 1 = 12MHz
    clock_configure(clk_ref,
                    CLOCKS_CLK_REF_CTRL_SRC_VALUE_XOSC_CLKSRC,
                    0,  // No aux mux
                    12 * MHZ,
                    12 * MHZ);

    // CLK SYS = PLL SYS (125MHz) / 1 = 125MHz
    clock_configure(clk_sys,
                    CLOCKS_CLK_SYS_CTRL_SRC_VALUE_CLKSRC_CLK_SYS_AUX,
                    CLOCKS_CLK_SYS_CTRL_AUXSRC_VALUE_CLKSRC_PLL_SYS,
                    freq, freq);

    clock_configure(clk_peri,
                    0,  // Only AUX mux on ADC
                    CLOCKS_CLK_PERI_CTRL_AUXSRC_VALUE_CLKSRC_PLL_USB,
                    48 * MHZ,
                    48 * MHZ);
  }
}
bool check_sys_clock_khz(uint32_t freq_khz, uint *vco_out, uint *postdiv1_out, uint *postdiv_out) {
  uint crystal_freq_khz = clock_get_hz(clk_ref) / 1000;
  for (uint fbdiv = 320; fbdiv >= 16; fbdiv--) {
    uint vco = fbdiv * crystal_freq_khz;
    if (vco < 400000 || vco > 1600000) continue;
    for (uint postdiv1 = 7; postdiv1 >= 1; postdiv1--) {
      for (uint postdiv2 = postdiv1; postdiv2 >= 1; postdiv2--) {
        uint out = vco / (postdiv1 * postdiv2);
        if (out == freq_khz && !(vco % (postdiv1 * postdiv2))) {
          *vco_out = vco * 1000;
          *postdiv1_out = postdiv1;
          *postdiv_out = postdiv2;
          return true;
        }
      }
    }
  }
  return false;
}
static inline bool set_sys_clock_khz(uint32_t freq_khz, bool required) {
  uint vco, postdiv1, postdiv2;
  if (check_sys_clock_khz(freq_khz, &vco, &postdiv1, &postdiv2)) {
    set_sys_clock_pll(vco, postdiv1, postdiv2);
    return true;
  } else if (required) {
    panic("System clock of %u kHz cannot be exactly achieved", freq_khz);
  }
  return false;
}
```

Dresch123, 2021-09-07:
Tim,
Thanks so much! That works great. I ran CoreMark twice, once with a 125MHz system clock: 146.34 CoreMarks, 
I then ran it at 250MHz: 292.68 CoreMarks, just like you would expect.
I also ran the Bench Mark Test for SDFat and got a 42% increase in SD card read rate at the higher clock!
Awesome! Thank you again, really appreciate it. Bill

closed 2021-09-22 by facchinm