arduino/ArduinoCore-mbed #310: Cannot create a variable called "cmd"

moose4lord, 2021-08-30:
I found something weird. My sketch, which has a variable called "cmd", will not compile on the pi pico.
It throws this compile error:
.../Arduino15/packages/arduino/hardware/mbed_rp2040/2.4.1/variants/RASPBERRY_PI_PICO/libs/libmbed.a(ns_cmdline.o):(.bss.cmd+0x0): multiple definition of `cmd'

Even this simple sketch fails:
```c++
char cmd;

void setup() {
}

void loop() {
}
```

facchinm, 2021-09-13:
Hi @moose4lord ,
thanks for reporting! The issue is due to https://github.com/arduino/mbed-os/blob/extrapatches-6.12.0/features/frameworks/mbed-client-cli/source/ns_cmdline.c#L358-L364 local variables being spilled in `libmbed` library.
@pennam and I will file a PR to mbed-os to make tha variables static to prevent it polluting the namespace.

facchinm, 2021-09-29:
Fixed by #331

closed 2021-09-29 by facchinm