arduino/ArduinoCore-mbed #312: Missing USB Gamepad and Joystick support

milador, 2021-09-01:
There's no gamepad or joystick support with current code. It's used in many applications and I am wondering it's under development or if it's in your plans to be added.

facchinm, 2021-09-13:
Hi @milador,
there's no plan right now but the infrastructure is there https://github.com/arduino/ArduinoCore-mbed/tree/master/libraries/USBHID , so HID based libraries can be added very easily (you can check [this](https://github.com/facchinm/USBJoystick) working library for a very rough but functional Joystick)

closed 2021-09-13 by facchinm

milador, 2021-09-14:
@facchinm Thanks for putting this together. I managed to make a simple joystick and not as complete as what you got. The only issue is the CDC port that I need to disable it somehow as it confuses the host device. Thanks Again.

milador, 2021-09-14:
@facchinm I am hoping to open another issue but you might know the answer. Is Bluetooth HID mouse feasible with current core? I have not been able to figure out if the implementation is there. Thanks

facchinm, 2021-09-15:
> The only issue is the CDC port that I need to disable it somehow as it confuses the host device.

Windows should trigger a driver reinstallation if the USB device descriptor has changed, but anyway disabling CDC makes sense in this case if it's the only thing you need to support

> Is Bluetooth HID mouse feasible with current core?

BLE HID requires security being implemented, and it's not yet avaialble in ArduinoBLE. However, on a Portenta and Nano33BLE, you can use mbed Cordio APIs. This library(https://github.com/tcoppex/mbed-ble-hid) wraps them in a library. Let me know if it works :wink:

milador, 2021-09-15:
@facchinm My goal is to make it work with Xbox Adaptive Controller which is very strict about the descriptor and I have gone through different methods of making it work with M0 and M4 base development board includeling modifying to the bootloader and generating uf2 file. Now drag and dropping a new uf2 file would work but it needs to be maintained and does not work out of the box for average maker.

This library seems to only work with Nano 33 BLE and not Nano rp2040. I started to look into porting it for Nano rp2040 to make it functional for my use case. The ble.h doesn't seem be available for rp2040.Any suggestions or tips will be appreciated.