arduino/ArduinoCore-mbed #318: RP2040 Connect board has faulty components in newest purchase

iKK001, 2021-09-07:
Using 8 out of the box new `Arduino Nano RP2040 Connect` boards, I have a problem with all of them concerning the library [WiFiManager_NINA_Lite](https://github.com/khoih-prog/WiFiManager_NINA_Lite).

Two months ago I already purchases an `Arduino Nano RP2040 Connect` board - and this one works nicely with the above library.

Now I think that there is a different Flash-memory component (or other) on the newly purchased faulty boards.

I am already in contact with Arduino Store to return those 8 faulty boards.

But maybe somebody from the ArduinoCore-mbed team has an idea what is going on.

Running the following default-Example I get the following Error message:

--> Example --> WiFiManager_NINA_Lite --> RP2040_WIFININA  :

```
Starting RP2040_WiFiNINA on MBED NANO_RP2040_CONNECT
WiFiManager_NINA_Lite v1.4.0
[WN] Hostname=RP2040-Master-Controller
LittleFS size (KB) = 64
LittleFS Mount Fail
Formatting... 
```

Or running the following two default-Examples I get the same Error message:

--> Example --> LittleFS_Mbed_RP2040 --> LittleFS_Counting

```
Start LittleFS_Counting on Nano RP2040 Connect
LittleFS_Mbed_RP2040 v1.0.2
[LFS] LittleFS size (KB) = 64
[LFS] LittleFS Mount Fail
[LFS] Formatting... 
```

or respectively,
-->  Example --> LittleFS_Mbed_RP2040 --> LittleFS_Test

```
Start LittleFS_Test on Nano RP2040 Connect
LittleFS_Mbed_RP2040 v1.0.2
[LFS] LittleFS size (KB) = 64
[LFS] LittleFS Mount Fail
[LFS] Formatting... 

```

So same behaviour for every Example **"LittleFS Mount Fail"**

-------------------------------------------
Here is my config:

I install the Arduino Mbed OS Nano Boards (v2.4.1) :

I installed the following Arduino libraries:
[image]

The selected Board is the Arduino Mbed OS Nano Boards-->Arduino Nano RP2040 Connect
(as the official Arduino install instructions for this board type suggest)

[image]

I installed the following Arduino libraries:

[image]

and 

[image]

I use the newest Arduino IDE (hourly build from yesterday) - but also the official release Arduino IDE 1.8.16 does not work.

[image]

Below are images of the old purchase board and one image of the new batch of boards. They are in a terrible state ! (and I already complained at the official Arduino Store to sell such awfully looking boards). The RP2040 is supposed to be a shiny example of IOT !

First the 2-month old working board:

[image]

And here one of the 8 awfully looking new boards that do not work ! 

[image]

pennam, 2021-09-08:
I've got a board with the adesto flash and the example sketch of  LittleFS_Mbed_RP2040 library is failing with an mbedOS Hard Fault. It's likely a software issue, but need some time to investigate.

pennam, 2021-09-09:
@iKK001 i'va made a couple of test and found out that the example is working with  [core 2.3.1](https://github.com/arduino/ArduinoCore-mbed/releases/tag/2.3.1) Could you plese give it a try on your board?

You can easily downgrade the core from the IDE 
[image]

iKK001, 2021-09-09:
Nice finding, and Yes, I will make the tests tonight (> 20:00h - since no time before that). I will get back to you as soon as have it running with 2.3.1 core, ok?

iKK001, 2021-09-09:
@pennam  : Thank you very much Mattia, for your finding!

**As for my "faulty boards", the WiFiManager_NINA_Lite is now working with the older mbed-core version 2.3.1.**

I first de-installed the version 2.4.1 with the Board-Manager in the Arduino-IDE.

And then I installed the previous version 2.3.1 - and now everything works again !

[image]

Starting the WiFiManager_NINA_Lite, I now get the correct terminal printout as follows:

```
LittleFS size (KB) = 64
LittleFS Mount OK
LittleFS Flag read = 0xd0d04321
```

Do you know where to address this issue  with the newest board version 2.4.1 ??

pennam, 2021-09-10:
@iKK001 Thanks for testing this on your boards. Don't know yet which is the root cause of the issue, but we will take care of it. I will update this issue as soon as i have a solution.