arduino/ArduinoCore-mbed #321: Portenta H7 UDP Multicast not working - mbed v2.4.1

Zuroy, 2021-09-13:
UDP Multicast is still not working in this version.

facchinm, 2021-09-13:
Hi @Zuroy,
does it fail to compile or "simply" doesn't work?
Can you provide the code you are using?
Thanks

Zuroy, 2021-09-13:
I'm trying to use the MDNS_generic Library which uses UDP Multicast. The code il correctly compiled but it does not work.

facchinm, 2021-09-13:
AFAIU `MDNS_generic` doesn't work out of the box for the Portenta since it requires `WiFiNINA_generic` ; did you patch it somehow?

Zuroy, 2021-09-13:
Nope. Do you have any suggestion?

khoih-prog, 2021-09-13:
Already try with `ArduinoCore-mbed` mbed_portenta core from v2.0.0 up to v2.4.1 without success, with or without the patches in [**Portenta_Patches**](https://github.com/khoih-prog/MDNS_Generic/tree/master/LibraryPatches/Portenta).

The Patches are copied / modified from this core's PR [**WiFi UDP & Multicast Functionality** #14](https://github.com/arduino/ArduinoCore-mbed/pull/14)

Zuroy, 2021-09-27:
Do you have any news? I'm trying to rewrite the C code but I'm stuck on the phy configuration.

facchinm, 2021-09-28:
We allocated some time to look for a fix in our backlog, hope it will be fixed by the next core release :slightly_smiling_face:

Zuroy, 2021-09-28:
> We allocated some time to look for a fix in our backlog, hope it will be fixed by the next core release 🙂

Thank you very much!

facchinm, 2021-09-28:
@Zuroy @khoih-prog I pushed #337 , tested it with Portenta, ArduinoMDNS and WiFi and it works fine :tada: 
On Ethernet, these lines should be added to the sketch after `Ethernet.begin()` (I'll integrate them in the next core revision)

```
  STM32_EMAC &emac = STM32_EMAC::get_instance();

  ETH_MACFilterConfigTypeDef pFilterConfig;
  HAL_ETH_GetMACFilterConfig(&emac.EthHandle, &pFilterConfig);
  pFilterConfig.PassAllMulticast = ENABLE;
  HAL_ETH_SetMACFilterConfig(&emac.EthHandle, &pFilterConfig);
  ```

@manchoz @pennam

khoih-prog, 2021-09-28:
Hi @facchinm , @Zuroy , @pennam , @manchoz 

I've just tested and confirmed the PR [**SocketWrapper: UDP: send packet on endPacket(), not on write()** #337](https://github.com/arduino/ArduinoCore-mbed/pull/337) fixes the UDP Multicast issue.

Many thanks for the beautiful fix

BR,

---

Test Results

1. Using Portenta_H7 WiFi

1.1 Terminal output

```
Start WiFiRegisteringServices on PORTENTA_H7_M7
MDNS_Generic v1.3.0-b1
Attempting to connect to SSID: HueNet1
SSID: HueNet1
Local IP address: 192.168.2.94
Signal strength (RSSI):-35 dBm
Registering mDNS hostname: portenta-h7-m7
To access, using portenta-h7-m7.local
[MDNS] ::begin: UDP beginMulticast statusCode= OK
[MDNS] ::_sendMDNSMessage: xid= 0
[MDNS] ::_sendMDNSMessage: queryCount= 0
[MDNS] ::_sendMDNSMessage: answerCount= 1024
[MDNS] ::_sendMDNSMessage: authorityCount= 0
[MDNS] ::_sendMDNSMessage: additionalCount= 256
[MDNS] ::_sendMDNSMessage: statusCode = 1
```
1.2 Ping using mDNS hostname => OK

```
kh@kh-Inspiron-3593:~$ ping portenta-h7-m7.local
PING portenta-h7-m7.local (192.168.2.94) 56(84) bytes of data.
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=1 ttl=255 time=49.4 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=2 ttl=255 time=153 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=3 ttl=255 time=73.4 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=4 ttl=255 time=95.7 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=5 ttl=255 time=119 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=6 ttl=255 time=22.0 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=7 ttl=255 time=30.4 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=8 ttl=255 time=47.7 m
```

2. Using Portenta_H7 Ethernet

2.1 Terminal output

```
Start RegisteringServices on PORTENTA_H7_M7 using Ethernet using Portenta_Ethernet Library
MDNS_Generic v1.3.0-b1
=========================
Default SPI pinout:
MOSI:8
MISO:10
SCK:9
SS:7
=========================
[MDNS] Board : PORTENTA_H7_M7 , setCsPin: 10
=========================
Currently Used SPI pinout:
MOSI:8
MISO:10
SCK:9
SS:7
=========================
Using mac index = 11
Connected! IP address: 192.168.2.87
Registering mDNS hostname: portenta-h7-m7-15
To access, using portenta-h7-m7-11.local
[MDNS] ::begin: UDP beginMulticast statusCode= OK
AddService : PORTENTA_H7_M7_mDNS_Webserver._http
[MDNS] ::_sendMDNSMessage: xid= 0
[MDNS] ::_sendMDNSMessage: queryCount= 0
[MDNS] ::_sendMDNSMessage: answerCount= 1024
[MDNS] ::_sendMDNSMessage: authorityCount= 0
[MDNS] ::_sendMDNSMessage: additionalCount= 256
[MDNS] ::_sendMDNSMessage: statusCode = 1

```

2.2 Ping using mDNS hostname => OK

```
kh@kh-Inspiron-3593:~$ ping portenta-h7-m7-15.local
PING portenta-h7-m7-15.local (192.168.2.87) 56(84) bytes of data.
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=1 ttl=255 time=0.964 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=2 ttl=255 time=0.216 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=3 ttl=255 time=1.07 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=4 ttl=255 time=4.40 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=5 ttl=255 time=0.314 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=6 ttl=255 time=0.635 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=7 ttl=255 time=0.996 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=8 ttl=255 time=4.47 ms
```

closed 2021-09-29 by facchinm