arduino/ArduinoCore-mbed #323: Support for NVIC interrupts

prugit1, 2021-09-14:
Does this repository support NVIC interrupts in Portenta H7?

khoih-prog, 2021-09-16:
You can try and use this new [**Portenta_H7_TimerInterrupt** Library](https://github.com/khoih-prog/Portenta_H7_TimerInterrupt)

---

### Features

This library enables you to use Interrupt from Hardware Timers on an **STM32H7-based Portenta_H7** board. 

As **Hardware Timers are rare, and very precious assets** of any board, this library now enables you to use up to **16 ISR-based Timers, while consuming only 1 Hardware Timer**. Timers' interval is very long (**ulong millisecs**).

Now with these new **16 ISR-based timers**, the maximum interval is **practically unlimited** (limited only by unsigned long miliseconds) while **the accuracy is nearly perfect** compared to software timers. 

The most important feature is they're ISR-based timers. Therefore, their executions are **not blocked by bad-behaving functions / tasks**. This important feature is absolutely necessary for mission-critical tasks.

prugit1, 2021-09-16:
@khoih-prog Thank you for your reply.

I could not find this library in the library manager. I would like to know when it will be available.

Also, does this library supports to use of more than 1 timer?

khoih-prog, 2021-09-16:
It'll be listed within several hours as it's totally brand-new. You can download it directly via GitHub at [Portenta_H7_TimerInterrupt](https://github.com/khoih-prog/Portenta_H7_TimerInterrupt).

The library can support many timers, check [Available Timers for Portenta_H7](https://github.com/khoih-prog/Portenta_H7_TimerInterrupt#1-ok-to-use), such as

**TIM1, TIM4, TIM7, TIM8, TIM12, TIM13, TIM14, TIM15, TIM16, TIM17**

khoih-prog, 2021-09-16:
The delay happens because of [[**Incident**] issues with Library Manager indexer infrastructure #483](https://github.com/arduino/library-registry/issues/483)

prugit1, 2021-09-16:
@khoih-prog Thank you for your reply.

I have tested your example codes and all are working fine with Portenta H7. The example titled **"Argument_None"** pretty well satisfies my requirement.

But, I am still curious to know,  What does TIM_HandleTypedef *htim refer to in the below command?  

HAL_TIM_IRQHandler(TIM_HandleTypeDef *htim)

closed 2021-09-22 by facchinm