arduino/ArduinoCore-mbed #324: Question: How do I use hardware JPEG codec on Portenta H7

tokiedokie, 2021-09-14:
Portenta H7 has hardware JPEG codec. However I cannot find any information or example code that encrypt/decrypt jpeg format for arduino.
How do I encrypt/decrypt jpeg format using hardware JPEG codec?

facchinm, 2021-09-22:
Hi @tokiedokie ,
there is no ready-made Arduino example since we've not been using that functionality yet.
However, you can use the awesome openmv wrapper (https://github.com/openmv/openmv/blob/a7e0e23bce73e3a7db4ede392fc4c3ea4279a9da/src/omv/imlib/jpeg.c#L962-L1275) with very little modifications.
Just keep in mind that if you include functions in an Arduino sketch you must prepend the functions with `extern "C"`, otherwise they won't be linked and nothing will work :wink:

tokiedokie, 2021-10-12:
Hi @facchinm,
Thank you for your reply.
I read the code and I tried to implement to my code but I couldn't do it.
I will use a existing JPEG decode library at this time.

closed 2021-10-12 by tokiedokie

hpssjellis, 2023-02-08:
Just thought I would add to this closed Issue. While searching the stm32 documents at 

https://www.st.com/resource/en/datasheet/stm32h747xi.pdf

I found this about the hardware 

[image]

Curious what @bitbank2 the maker of [JPEGDEC](https://github.com/bitbank2/) and [JPEGENC](https://github.com/bitbank2/JPEGENC) has to say about the Portenta JPEG hardware. 

I have already used his [PNGenc](https://github.com/bitbank2/PNGenc) library to get PNG images saved to SD card on the portenta [here](https://github.com/hpssjellis/portenta-pro-community-solutions/blob/main/examples/dot3-portenta-vision-shields/dot36-camera-png-to-web/dot362-png-to-sd-card/dot362-png-to-sd-card.ino)

bitbank2, 2023-02-08:
The hardware JPEG codec of the STM32 is used in OpenMV. I optimized that code and the hardware outperforms software encoding/decoding if you feed it data efficiently. As @facchinm says, the best place to see how to use the STM32 JPEG hardware is in the OpenMV code.

metanav, 2023-10-10:
> Hi @tokiedokie , there is no ready-made Arduino example since we've not been using that functionality yet. However, you can use the awesome openmv wrapper [...]

I am interested in having the ArduinoCore Mbed include support for JPEG hardware encode/decode. I attempted to implement it using the OpenMV code or directly from the STM32CubeH7 examples, but they were unsuccessful with Mbed. With the introduction of the Giga R1 Display shield, direct support of this kind is necessary.

bitbank2, 2023-10-10:
@metanav if hardware JPEG decode isn't a 'hard' requirement, my JPEGDEC software decoder can work with any size image. The default buffer sizes are to allow it to run on devices with 20K SRAM or more.

metanav, 2023-10-10:
> @metanav If hardware JPEG decode isn't a 'hard' requirement, my JPEGDEC software decoder can work with any size image. The default buffer sizes are to allow it to run on devices with 20K SRAM or more.

Thanks, @bitbank2 for your reply. I needed hardware JPEG decoding to render fast-moving frames in RGB565 800x480 pixels at 25 fps. However, this seems implausible with software decoding.

bitbank2, 2023-10-10:
It may be possible, but it will be mostly dependent on the quality setting of the images. Large, high quality frames will decode more slowly than small low quality frames.